Klipper3d/klipper #5012: Cant connect to MCU after update

StefanMeGit, 2021-12-09:
Hi Klipper Friends,

i updated my system since a while and run into some major issues. At first i had a hard time to flash the ARDUINO DUE. The flash process stopped randomly. I figured out, that Octoprint is continously try to reconnect to the DUE and interrupt the flash process. I stopped the octoprint service... flashing worked like a charm.
[image]

After restaring the RPI, i still cannt connect to the MCU. Any ideas what i am doing wrong? The system worked before the update just fine.
[klippy.log](https://github.com/Klipper3d/klipper/files/7683804/klippy.log)

Sineos, 2021-12-09:
Verify if your serial path
```
[mcu]
serial = /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55839313638351F041C2-if00
```
is still correct.

If yes:
1. Repeat the flash process
2. Make sure to power down the board completely
3. Reattach board to the Pi / power up
4. Post output of `ls /dev/serial/by-id/*` command
4. Post output of `dmesg` command
5. Post output of `lsusb` command

StefanMeGit, 2021-12-09:
Thanks for the help. 

just did everything u mentioned exepct of power off an on the arduino (just not at  home at the moment... did everything by remote).  I ll do it as soon as i get home. 

The serial looks correct.
[image]

```
pi@octopi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1496 SMP Wed Dec 1 15:58:11 GMT 2021
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x34000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] On node 0 totalpages: 229376
[    0.000000]   DMA zone: 2016 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 229376 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=512 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=874acb7a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 826028K/917504K available (10240K kernel code, 1311K rwdata, 2948K rodata, 1024K init, 859K bss, 25940K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32054 entries in 63 pages
[    0.000000] ftrace: allocated 63 pages with 6 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000315] Console: colour dummy device 80x30
[    0.001174] printk: console [tty1] enabled
[    0.001241] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001299] pid_max: default: 32768 minimum: 301
[    0.001522] LSM: Security Framework initializing
[    0.001780] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001829] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003308] cgroup: Disabling memory control group subsystem
[    0.003596] CPU: Testing write buffer coherency: ok
[    0.004114] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005374] Setting up static identity map for 0x100000 - 0x10003c
[    0.005583] rcu: Hierarchical SRCU implementation.
[    0.006528] smp: Bringing up secondary CPUs ...
[    0.007706] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008989] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.010307] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[... 243 lines cut ...]
[    8.281258] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.338096] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[   10.027368] random: crng init done
[   10.027385] random: 7 urandom warning(s) missed due to ratelimiting
[   10.354844] uart-pl011 3f201000.serial: no DMA platform data
[   10.461774] 8021q: 802.1Q VLAN Support v1.8
[   10.464279] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   10.994609] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[... cut ...]
[   17.712425] Bluetooth: HCI UART protocol H4 registered
[   17.712466] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   17.712533] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.712763] Bluetooth: HCI UART protocol Broadcom registered
[   18.057254] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   18.057269] Bluetooth: BNEP filters: protocol multicast
[   18.057296] Bluetooth: BNEP socket layer initialized
```

Sineos, 2021-12-09:
I never used a Due with Klipper and I'm surely not an expert at it, but IIRC correctly the Due has two USB ports

`[    4.625515] usb 1-1.1.3: Product: Arduino Due Prog. Port`

sounds like the programming port. Does the other USB port make a difference?

StefanMeGit, 2021-12-18:
Thanks, i found something out.

It looks like there was a big change in the past. I was able to connect to the MCU via the other port. 

I used the Prog. Port for 3 years. After the update i had to use the SamX dirct chip port to connect.... very strange. I also had to reconfigure all the PINS, because they have now different names.

Now i run into another issue.

For some reason my Prog. Port changes from time to time.

i got it working at the time the Pi recognice the port as:
```
pi@octopi:~ $ ls /dev/serial/by-id/*
serial: /dev/serial/by-id/usb-Klipper_sam3x8e_20312051513348323136303039303132-if00
```

but after a restart the usb name changed to:
```
pi@octopi:~ $ ls /dev/serial/by-id/*
serial: /dev/serial/by-id/usb-03eb_6124-if00
```
i tried to connect now via `serial: /dev/serial/by-id/usb-03eb_6124-if00` which doesnt work again. I need a frew restarts or some other random stuff to get the old name like serial: `/dev/serial/by-id/usb-Klipper_sam3x8e_20312051513348323136303039303132-if00` back.

very strange, Does anyone have a hint whats going on here?

thanks a lot

closed 2022-01-06 by StefanMeGit