Klipper3d/klipper #5016: Move exceeds maximum extrusion - Only on consecutive Print

theely, 2021-12-11:
Any time I try to start a second print without restarting Klipper first I run into exceeds maximum extrusion.
The first print works always fine, if however it try to re-print the same file (gcode) or a different file the print halts and the error message maximum extrusion is displayed.

Error message:
12:58:25 
// Move exceeds maximum extrusion (9.804mm^2 vs 0.640mm^2)
// See the 'max_extrude_cross_section' config option for details
12:58:25 
!! Move exceeds maximum extrusion (9.804mm^2 vs 0.640mm^2)

Attached a log file, first print was ok second attempt run into issue:
[klippy.log.1.zip](https://github.com/Klipper3d/klipper/files/7697073/klippy.log.1.zip)

Gcode:
[Shell Front (1)_4h36m_0.2mm_215C_PLA.gcode.zip](https://github.com/Klipper3d/klipper/files/7697084/Shell.Front.1._4h36m_0.2mm_215C_PLA.gcode.zip)

Sineos, 2021-12-12:
Check your start / stop gcode and slicer settings. Most likely you have some `M83` commands causing your extrusion to add up

theely, 2021-12-12:
The start code does not make use of M83, however after changing from absolute to relative positioning the issue was gone.
The amount of extruded material has not changed, I am not sure what is going on.

Absolute Positioning Start G-code (problematic, Move exceeds maximum extrusion on consecutive prints)

M117 Prime nozzle.
G92 E0                                            ; Reset Extruder
G1 X20 Y4 Z2 F5000                     ; Move to start position
G1 X150 Y4 Z0.3   F1500.0 E15    ; Draw the first line
G1 X150 Y4.4 Z0.3 F5000.0         ; Move to side a little
G1 X20  Y4.4 Z0.3 F1500.0 E30   ; Draw the second line
G1 X20  Y10  Z5   F5000   E29.4  ; Quickly move away
G92 E0                                           ; Reset Extruder

Absolute Positioning Start G-code (no issue on consecutive prints)
M117 Prime nozzle.
G92 E0                                          ; Reset Extruder
G1 X20 Y4 Z2 F5000                   ; Move to start position
G91                                                ; Relative positioning
G1 X130  Y0   Z0 F1000.0 E15    ; Draw the first line
G1 X0      Y0.4 Z0 F5000.0         ; Move to side a little
G1 X-130 Y0.4 Z0 F1000.0 E15  ; Draw the second line
G1 X0       Y6   Z5 F5000   E-0.6; Quickly move away
G92 E0                                         ; Reset Extruder
G90                                              ; Absolute positioning

Sineos, 2021-12-30:
Issue seems concluded. If you need support, we will be happy to assist on https://klipper.discourse.group

closed 2021-12-30 by Sineos