Klipper3d/klipper #5023: processing macro commands

legend069, 2021-12-14:
i have noticed that when klipper process a macro it does it "all at once" loading into a buffer then executes each command.

this however can be problematic when i have a conditional statement that runs based if a temperature is met, due to the way klipper works and the buffer.
it will get the temperature before it is suppose to.

same thing happens when you have a check if homed command
followed by another check if homed command 
it will check twice and both times will be false.

I've uploaded my macro I've been having issues with.

[scrubber_v2.zip](https://github.com/Klipper3d/klipper/files/7711078/scrubber_v2.zip)

Sineos, 2021-12-14:
See https://klipper.discourse.group/t/macro-creation-tutorial/30?u=sineos and especially  https://klipper.discourse.group/t/macro-creation-tutorial/30/6?u=sineos

jakep82, 2021-12-14:
This is also noted in the official Klipper docs.

`Important! Macros are first evaluated in entirety and only then are the resulting commands executed. If a macro issues a command that alters the state of the printer, the results of that state change will not be visible during the evaluation of the macro. This can also result in subtle behavior when a macro generates commands that call other macros, as the called macro is evaluated when it is invoked (which is after the entire evaluation of the calling macro).
`

https://www.klipper3d.org/Command_Templates.html#the-printer-variable

closed 2021-12-29 by legend069