Klipper3d/klipper #5024: Octopus Pro with onboard MAX31865 - temperature issue

Afterschnitzel, 2021-12-14:
Hello all, I am facing weird behaviour of MAX31865 on board with PT1000 thermistor on hotend.
When not printing, temperature is stable and heater heating is stable too. But if I start to print, temperature of hotend is jumping +-2 degrees and heater is heating jumping up and down (my part cooling is turned off).
Thermistor is OK, it works normally when I connect to T0 terminal.

Any suggestions, please?

Here is mi configuration of MAX31865 in extruder section:

```
sensor_type: MAX31865
sensor_pin: PF8
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
rtd_nominal_r: 100
rtd_num_of_wires: 2
rtd_reference_r: 430
```
[image]
[klippy.log](https://github.com/Klipper3d/klipper/files/7711995/klippy.log)

Sineos, 2021-12-14:
PT100 or PT1000?

Afterschnitzel, 2021-12-14:
I wrote I have PT1000.
Please read carefully.

Sineos, 2021-12-14:
Well, if it is a PT1000, then you should probably set the `rtd_nominal_r` --> Please read carefully
Also MAX31865 boards are usually different for PT100 and PT1000. Typically PT1000 MAX boards have a `rtd_reference_r` of 4300 --> Please read carefully
And if your power grid has a 50Hz frequency, you should set `rtd_use_50Hz_filter: True` --> Again, you know, please read ...

Afterschnitzel, 2021-12-14:
> Well, if it is a PT1000, then you should probably set the `rtd_nominal_r: 1000` --> Please read carefully Also MAX31865 boards are usually different for PT100 and PT1000. Typically PT1000 MAX boards have a `rtd_reference_r` of 4300 --> Please read carefully And if your power grid has a 50Hz frequency, you should set `rtd_use_50Hz_filter: True` --> Again, you know, please read ...

No.
If i use `rtd_nominal_r: 1000` temperature shows -200°C on ambient temperature
and `rtd_use_50Hz_filter: True` has no impact on this behaviour.

Afterschnitzel, 2021-12-14:
No change with this settings:

[image]

[image]

Sineos, 2021-12-14:
https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier
https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/python-circuitpython#python-installation-of-max31865-library-2999057-6

[image]

Sineos, 2021-12-14:
Maxim MAX31865 Datasheet: https://datasheets.maximintegrated.com/en/ds/MAX31865.pdf
[image]

Afterschnitzel, 2021-12-14:
Let me see the Octopus Pro scheme - 4300 Ohms.
But it can't solve my problem.
[image]

Sineos, 2021-12-14:
Sorry, no further idea. I assume you have configured your onboard MAX correctly to use 2 wires and the appropriate sensor resistance. Also your Klipper settings seem to be correct now.

Somehow looks like there is suddenly "noise" disturbing the sensor readings when the print starts. Could be some EMI effects, like sensor cable running along the stepper cables etc. Could be poor board design by BTT, although it looks like they have taken measures to clean the VCC (RL circuit)

What I see on my MAX boards is a drop in temperature as well, when the extrusion starts but then the temperature quickly stabilizes and stays constant.

Afterschnitzel, 2021-12-15:
Thanks for reply.
Tha last thing I have to try is disconnect 60V power supply for XY steppers. Temperature problem obviously happen when motors are moving.

Sineos, 2022-01-07:
Seems the topic is concluded. Please do not hesitate to open a new issue on https://klipper.discourse.group if needed

closed 2022-01-07 by Sineos

hapklaar, 2022-01-15:
@Afterschnitzel did you manage to fix your pt1000 problem?

mlauber, 2022-03-27:
> Hello all, I am facing weird behaviour of MAX31865 on board with 2-wire PT1000 thermistor on hotend. When not printing, temperature is stable and heater heating is stable too. But if I start to print, temperature of hotend is jumping +-2 degrees and heater is heating jumping up and down (my part cooling is turned off). Thermistor is OK, it works normally when I connect to T0 terminal.
> 
> Any suggestions, please?
> 
> Here is mi configuration of MAX31865 in extruder section:
> 
> ```
> sensor_type: MAX31865
> sensor_pin: PF8
> spi_software_sclk_pin: PA5
> spi_software_mosi_pin: PA7
> spi_software_miso_pin: PA6
> rtd_nominal_r: 100
> rtd_num_of_wires: 2
> rtd_reference_r: 430
> ```
> 
> [image]
> 
> [klippy.log](https://github.com/Klipper3d/klipper/files/7711995/klippy.log)

Did you ever get this figured out?  I have an BTT Octopus Pro and am struggling to get my PT100 to work.  I do not have an amplifier board, but I don't know if that the problem or not.  I read that the octopus pro should accept the PT100 without one, but when I follow their instructions I get some crazy 1093 degree C reading the octoprint shuts down as a result.  

I'd really appreciate if someone could help me out.

visilya, 2022-06-01:
heater_pin = PA2
sensor_type = MAX31865
sensor_pin = PF8
spi_speed = 4000000
#spi_bus: spi1
spi_software_miso_pin = SPI1_MISO
spi_software_mosi_pin = SPI1_MOSI
spi_software_sclk_pin = SPI1_SCK
rtd_nominal_r = 1000
rtd_reference_r = 4300
rtd_num_of_wires = 2
rtd_use_50Hz_filter = True
min_temp = 10
max_temp = 350
max_power = 1.0
min_extrude_temp = 170

Connected by coupling left and right pairs: first wire - 1+2 and second wire - 3+4

CD1010, 2022-11-17:
You have to set dip pins according to manual also.