Klipper3d/klipper #5025: Problem with color change with M600 macro, print doesn't stop and continues printing

jccf07, 2021-12-15:
**My problem**

Hi,  I have a problem when i make filament changes with the macro M600, because the printing does not stop and continues printing until the end. I use recommend configuration in documentation files 
It's a bit strange, because this behavior only happens sometimes, about 7 out of 10 prints are ok. The M600 acts ok (it pauses print, I change the filament a few hours later, y send RESUME command and it continues perfectly).
Other times, print doesnt stop and printing continues to the end without pausing

I think it could be a problem with macros definition, but I don´t know how to solve 

I attach fluidd console output
[image]

**My related macros:**

[pause_resume]

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
# change this if you need more or less extrusion
variable_extrude: 1.0
gcode:
  ##### read E from pause macro #####
  {% set E = printer["gcode_macro PAUSE"].extrude|float %}
  ##### set park positon for x and y #####
  # default is your max posion from your printer.cfg
  {% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
  {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
  ##### calculate save lift position #####
  {% set max_z = printer.toolhead.axis_maximum.z|float %}
  {% set act_z = printer.toolhead.position.z|float %}
  {% if act_z < (max_z - 2.0) %}
      {% set z_safe = 2.0 %}
  {% else %}
      {% set z_safe = max_z - act_z %}
  {% endif %}
  ##### end of definitions #####
  PAUSE_BASE
  G91
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G1 E-{E} F2100
  {% else %}
    {action_respond_info("Extruder not hot enough")}
  {% endif %}
  {% if "xyz" in printer.toolhead.homed_axes %}
    G1 Z{z_safe} F900
    G90
    G1 X{x_park} Y{y_park} F6000
  {% else %}
    {action_respond_info("Printer not homed")}
  {% endif %} 

[gcode_macro M600]
gcode:
    {% set X = params.X|default(50)|float %}
    {% set Y = params.Y|default(0)|float %}
    {% set Z = params.Z|default(10)|float %}
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    G91
    G1 E-.8 F2700
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
    G91
    G1 E-100 F1000
    RESTORE_GCODE_STATE NAME=M600_state
    

**My configuration:**
Raspberry Pi 4 Model B Rev 1.2
Running: FluiddPI v1.16.2
fluidd v1.16.2
klipper v0.10.0-180-g323268ea
moonraker v0.7.1-243-g3f7cc53
Prusa Slicer 2.3.3 (using change color feature)

**Other info**
I attach my printer.cfg
[printer.cfg.txt](https://github.com/Klipper3d/klipper/files/7715489/printer.cfg.txt)
.

thghca, 2021-12-15:
Try CLEAR_PAUSE at end of Resume and cancel macros.

jccf07, 2021-12-15:
I have noticed that this has always happened to me after several prints, I am leaving the printer on for many days in a row making prints. 
Sometimes due this error, I tried to change some things in the "printer.cfg" and I restarted, and then, after several prints worked for me and it happened again.

I hope this is the solution. To prevent, I will restart the printer + klipper + moonraker after each print. 
Also I will add the CLEAR_PAUSE command after RESUME and CANCEL_PRINT
Thanks, I will write a feedback if it solve the problem or not

ozanges, 2022-01-08:
Hi, 

I am having same problem here. 
Attached some data to analyze : gcode, klippy.log -> klipper v0.9.1... (very huge, a day of tests) & klipper v0.10.0... (some config errors on klipper upgrade & one test)

I have this issue at each test

Hardware : 
 - btt octopus (buy on august 2020)
 - rpi 4 
 - other stuff from voron 2.4 initial build

Versions : 
 - Superslicer 2.3.57.8
 - Octoprint : 1.7.2 on Octopi : 0.18.0
 - Klipper : v0.10.0-222-g7c0559c6-dirty  (and v0.9.1-740-g25e493ef)

What am i doing : 
 - slice test object and define 3 colors part
 - Set "PAUSE" into color change gcode
 - Sending gcode from Superslicer to Octoprint
waiting for first PAUSE : it works
send RESUME from Octoprint or LCD
print restart
waiting for second PAUSE : it does not work
 - no signs of PAUSE 
 - I can PAUSE and RESUME from LCD : it works
 - the second PAUSE from gcode is never triggered
 
tried some M600 macro found on the net, then just had [pause_resume] and finally add CLEAR_PAUSE after RESUME but does not work either

no macro in octoprint.
just found in logs : 
action:paused
...
action:resume
[data.zip](https://github.com/Klipper3d/klipper/files/7832138/data.zip)

...
Print already paused
...

i can't understand why i can PAUSE with LCD at anytime but not with sliced gcode object after one success. 

glad to try another things, understand my mistake or another data. 
thank you

jccf07, 2022-01-08:
Hi, PAUSE CLEAR did not solve my problem.  I am printing with 2 colors, the only way that everything goes well, is to restart the printer, the software and firmware before each print.  it's very strange.

Even if I just change the filament, extruding -100 mm and then +150 mm, the printer head goes some times to the Xmax Ymax point, then the printing does not stop on the M600.
For me, as I said, the only way that always works is to reboot the machine, the software and the firmware and then start printing without do not any command before.

It looks that's it is a klipper or Moonraker problem

mental405, 2022-01-08:
This is probably an issue that needs to be troubleshot on Discourse or Discord. If it is determined to be a bug in the code, we can re-open it here.

closed 2022-01-10 by github-actions[bot]