Klipper3d/klipper #5029: homing_override seems to override even the individual axis commands with G28 X et al.

redfox05, 2021-12-15:
This is referencing the feature added in #25.

Klippy Log attached
[klippy-log-20211215-2142.log](https://github.com/Klipper3d/klipper/files/7722875/klippy-log-20211215-2142.log)

## Problem/Actual Results
I noticed when using `[homing_override]`, the individual G28 X, G28 Y, G28 Z commands no longer just do a single axis now for me.

## Question/Expected Results
Is there a way to override the order, such that doing JUST G28 will trigger the new order, but keeping it as the supplied axis when G28 has an axis value passed to it?

## Config
My config for this section is as follows:
```
[homing_override]
axes: xyz
gcode:
  # G1 Z10 F6 # No need to do this, G28 Z already does it. Move Z up just incase the nozzle is too close to the bed before homing (can be the case in cancelled prints/powercuts/emergency stops)
  G28 Z # Home z, because detector is on the bed, not the nozzle
  G28 Y # Do Y next, to avoid potentially hitting the back of the nozzle wipe.
  G28 X
```

### Background context
On my printer (E3D BigBox Pro), I have mounted the Z probe on the bed, and it measures against a fixed shelf/point on the printer frame. And if it helps, the X axis (and motor) sits on the Y rods (and Y motor is on the frame).
Reason I want Z then Y then X, is to avoid the nozzle scraping along the bed in a sudden reset, and Y before X is because there is a cutout at the back of the frame to allow the extruder carriage to fully access the nozzle wipe/ooze box.
So I want it to move clear out of that to avoid hitting that cutout.

jakep82, 2021-12-16:
The override runs for any axis specified by `axes`. You've told it to run to the override whenever any axis is homed.

redfox05, 2021-12-16:
Okay, so how can I simply apply it only when G28 is run without params?

Sadly when this went in the pull request was not referenced to the original
issue creating it, otherwise I would have had a look at the code to see how
it was designed to work. And if it was a bug I could even suggest a PR.

But the way I see this, it should really only trigger on a G28 or G28 xyz

jakep82, 2021-12-16:
> Okay, so how can I simply apply it only when G28 is run without params? 

This is not possible. The purpose of the override is to safely home a printer that requires a particular sequence for a particular axis.  You must specify at least one axis, and the override will run whenever that axis is homed.  If you specify z, the override will only be run for G28 or G28 Z.  

> Sadly when this went in the pull request was not referenced to the original issue creating it, otherwise I would have had a look at the code to see how it was designed to work. And if it was a bug I could even suggest a PR. But the way I see this, it should really only trigger on a G28 or G28 xyz

This is not a bug.  If you need more information please read the config reference document, but this feature is working as intended.

<https://www.klipper3d.org/Config_Reference.html#homing_override>

redfox05, 2021-12-16:
> This is not a bug. If you need more information please read the config reference document, but this feature is working as intended.

Hi Jake, thanks for your reply, but I did check the docs, it did not mention the exact behaviour of when it applies, so it's not really too helpful in this case.

Appreciate you feel this is 'working as intended' but as with all things, it's good to call out when something doesn't seem quite right, and my point is valid, as nowhere explains why a standard G28 with no params should trigger the override. In fact, a previous PR for this issue mentioned in #25 actually suggested doing the way I'm discussing. There's lots of printers that need to have different axis homed first, and especially in emergency situations, one may need to move (but first home) a single axis without affecting the others. The current method would make it a little tricker.

IMO, I think it would be logical way to only apply the override to the axis mentioned in the `axes: ` config section, and only if all are mentioned as `axis: xyz` then also apply it to the general G28 home all command.

If you don't agree, please do explain why.

@KevinOConnor appreciate you're probably very busy, and this is low priority, but would you be able to point out where this was done in the code, and I could potentially submit a PR for discussion. Alas, thinking about it, as this feature is now released, other people may be relying on the way its currently implemented, so changing it might not be a good idea anyway.

### Potential workaround
In the meantime, in case it helps anyone else, I think the only solution would be to only apply the override to `axis: x` (or y) and then at least a G28 would get intercepted by the override, but a G28 Z would not, and only do Z.

simonkuehling, 2022-01-05:
@redfox05 the code is in the [`homing_override.py`](https://github.com/Klipper3d/klipper/blob/8a3727ef742f46923275ffca4651710952cfa114/klippy/extras/homing_override.py#L20) module. The G28 command is overwritten there.

What if you just removed your `[homing_override]` config section and instead created your own G28-rewrite with a macro like
```
[gcode_macro G28]
rename_existing: G28.1
gcode:
    ...
```
here you can craft the G28 to behave just like you need. Is that an option?

redfox05, 2022-01-05:
Ah interesting. I'll look into that thanks. As long as you can specify
behaviour for G28 on its own as well as parameters, then that might just
work.

Still seems a lot of workaround for something I imagined would be quite
common (home Z before others), but I guess not, others would have chimed in
by now.

simonkuehling, 2022-01-05:
Sure, should be as easy as this (untested):
```
[gcode_macro G28]
rename_existing: G28.1
gcode:
    {% if params.X and not params.Y and not params.Z %}
        G28.1 X
    {% endif %}
    {% if not params.X and params.Y and not params.Z %}
        G28.1 Y
    {% endif %}
    {% if not params.X and not params.Y and params.Z %}
        G28.1 Z
    {% endif %}
    {% if not params.X and not params.Y and not params.Z %}
        # G1 Z10 F6 # No need to do this, G28 Z already does it. Move Z up just incase the nozzle is too close to the bed before homing (can be the case in cancelled prints/powercuts/emergency stops)
        G28.1 Z # Home z, because detector is on the bed, not the nozzle
        G28.1 Y # Do Y next, to avoid potentially hitting the back of the nozzle wipe.
        G28.1 X
    {% endif %}
```

jakep82, 2022-01-05:
This workaround accomplishes what you want, but it's also potentially dangerous.  It will allow you to home X or Y regardless of whether Z has been homed, or the bed is safely clear of the nozzle.  That's why the homing_override works the way it does.  It eliminates the potential for user error.

redfox05, 2022-01-06:
Thanks both, for the example and the warning. Perhaps I should just stick
with how it is now. The likelihood of needing to home individual is
probably pretty low. I'd rather have it do it safely as you say.

closed 2022-02-17 by github-actions[bot]

Cyphon00, 2022-06-18:
I wasn't able to get @simonkuehling's code working in my application. It only worked when a 0 was included with the axis (eg: G28 X0 worked, but not G28 X).
That said, his work helped me build my own version. So thank you @simonkuehling !

For anyone interested in what I included in my version:
1. Works whether a 0 is included or not.
2. Forces the axes to home in the order X-Y-Z. (Important for some systems.)
3. Sets Absolute movement (G90) at start and end of code - change this if you need but I found it a good failsafe against people trying to move an unhomed axis using the GUI buttons +1,+10,+100.
4. Includes a retraction distance for the toolhead so it 'steps back' from the endstop - this reduces force on the endstop and allows the homing function to provide clearance for homing the next axis when using larger toolheads/probes.
5. Won't home Y unless X is homed and outside the clearance zone. (Similar to point 4, this was important to my setup to prevent the probe crashing into the endstop when homing Y).

Here's the code:

```
[gcode_macro G28]
rename_existing: G28.1 #Move the original G28 to G28.1
gcode:
    G90  #set to absolute XYZ mode.
    {% set min_safe_x = 30 %} #Define the minimum X value needed to safely do a Y home move, without crashing probe into gantry. This also provides the 'retact distance', so the toolhead doesn't keep pressing on the endstop.
    {% set min_safe_y = 10 %} #Define the minimum Y value needed to safely move print head off the endstop.  This also provides some 'retact distance', so the toolhead doesn't keep pressing on the endstop.

#For debugging purposes only (comment this section out after use): show current position, show XY mins, show final intended positions (based on min_safe x and y).
#NOTE: To use M118, Printer.cfg must include:
#   [respond]
#   default_type: echo
    #M118 Current Toolhead Position x:{printer.toolhead.position.x} y:{printer.toolhead.position.y}
    #M118 X axis min: {printer.toolhead.axis_minimum.x}, Y axis min: {printer.toolhead.axis_minimum.y}
    #M118 Resting Toolhead Position x:{min_safe_x} y:{min_safe_y}
#Debug code end

    {% if params.X is defined and params.Y is undefined and params.Z is undefined %} #Home X and move to safe location (min_safe_x mm) away from endstop.
        G28.1 X
        G1 X{min_safe_x} F10000
    {% endif %}

    {% if params.X is undefined and params.Y is defined and params.Z is undefined %} #Home Y and move to safe location (min_safe_y mm) away from endstop.
        
        {% if (printer.toolhead.homed_axes|truncate(1, False, '', 0)) != "x" or printer.toolhead.position.x < min_safe_x %} #If X isn't homed OR if X is within the safe_distance,  Home X before doing the Home Y.
		    G28.1 X			        #Home X and move to safe location (min_safe_x mm) away from endstop.
            G1 X{min_safe_x} F10000
            M118 used safe home X
	    {% endif %}
        
        G28.1 Y         #Here's where we continue to Home Y.
        G1 Y{min_safe_y} F10000
    {% endif %}

    {% if params.X is defined and params.Y is defined and params.Z is undefined %} #Home X and Y (in that order), moving each to their safe locations.
        G28.1 X
        G1 X{min_safe_x} F10000
        G28.1 Y
        G1 Y{min_safe_y} F10000
    {% endif %}

    {% if params.X is undefined and params.Y is undefined and params.Z is defined %} #Home Z - Klipper won't allow this to run if XY aren't homed, so I've not added any of my own protection code here.
        G28.1 Z
    {% endif %}

    {% if (params.X is defined and params.Y is defined and params.Z is defined) or (params.X is undefined and params.Y is undefined and params.Z is undefined)  %} #Home all axes (in order XYZ) if either G28 (no axes specified) or G28 XYZ (all axes specified) is called.
        G28.1 X
        G1 X{min_safe_x} F10000
        G28.1 Y
        G1 Y{min_safe_y} F10000
        G28.1 Z
    {% endif %}
    G90 #ensure absolute XYZ mode on completion.
##
```