Klipper3d/klipper #5032: Lost communication with MCU

twiget3dp, 2021-12-16:
I am continually getting this error on my V0.1. It happens about 15 minutes into every print. So far I've swapped out the SKR E3 V2 and the Pi 3B+. I've monitored the PSU output, at idle, it's rock steady at 24.00VDC, under load it drops to 23.98VDC and there's no change when the MCU error happens. All the wiring is good and I've tried 3 different USB cables and it still dies as the same spot with each cable. Any thoughts on what's going on?
[image]
[klippy.log](https://github.com/Klipper3d/klipper/files/7728866/klippy.log)

twiget3dp, 2021-12-16:
I found another post that indicated that if the power to the Pi isn't 100% stable it could cause this issue. I had been powering the Pi via the GPIO pins and a buck converter. I disconnected the converted and plugged the power supply that came with my Pi into the microUSB port and tried again. The printer died again, with the same error as before. The log is attached.
[klippy(1).log](https://github.com/Klipper3d/klipper/files/7729112/klippy.1.log)

Sineos, 2022-01-07:
If the issue still persists, please do not hesitate to open a new topic on https://klipper.discourse.group

closed 2022-01-07 by Sineos