Klipper3d/klipper #5033: Unknown pin chip name 'probe'

macaroniart123, 2021-12-17:
Hello, 

Im getting stuck at this point:
08:53:28 Klipper: Standby
08:53:32 Klipper: Connecting ...
08:53:32 Unknown pin chip name 'probe'

I am using teaching tech videos as a guide: https://www.youtube.com/watch?v=3DWqJeXK7yc

I read this part of klipper but do not think it has by answer?
https://www.klipper3d.org/BLTouch.html

Im following the 
Klipper log: https://pastebin.com/QB0ZTErV
Klipper config https://pastebin.com/C5j2sX35

Thank you!!

hmortensen83, 2021-12-17:
Your config sections safe_z_home, bltouch and bed_mesh are indented. Remove tabs and whitespaces infront of every line in the config.

Sineos, 2021-12-30:
Issue seems concluded. If you need support, we will be happy to assist on https://klipper.discourse.group

closed 2021-12-30 by Sineos