Klipper3d/klipper #5035: Error during first init after boot

Vittaly, 2021-12-17:
Hi All!

I have problem with first starting klippy after boot SKR 1.3.

I see in log:
```
mcu 'mcu': Starting serial connect
Starting stk500v2 leave programmer sequence
Got b'\x05\x13\xac\x1a~' from stk500v2
mcu 'mcu': Unknown message 82 (len 15) while identifying
mcu 'mcu': Unknown message 82 (len 15) while identifying
new minimum rtt 6357.704: hrtt=0.000113 freq=100000000
mcu 'mcu': got {'oid': 17, 'next_clock': 401840192, 'value': 31720, '#name': 'analog_in_state', '#sent_time': 6357.703886548, '#receive_time': 6357.749150881}
mcu 'mcu': got {'oid': 26, 'next_clock': 410840192, 'value': 31704, '#name': 'analog_in_state', '#sent_time': 6357.810853839, '#receive_time': 6357.839177589}
mcu 'mcu': got {'oid': 17, 'next_clock': 431840192, 'value': 31720, '#name': 'analog_in_state', '#sent_time': 6358.024695964, '#receive_time': 6358.049177006}
Loaded MCU 'mcu' 94 commands (v0.10.0-122-gc179db3d / gcc: (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] binutils: (2.34-4ubuntu1+13ubuntu1) 2.34)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c0=P0.28,P0.27 BUS_PINS_i2c1=P0.1,P0.0 BUS_PINS_i2c1a=P0.20,P0.19 BUS_PINS_i2c2=P0.11,P0.10 BUS_PINS_ssp0=P0.17,P0.18,P0.15 BUS_PINS_ssp1=P0.8,P0.9,P0.7 CLOCK_FREQ=100000000 MCU=lpc1768 PWM_MAX=255 RESERVE_PINS_USB=P0.30,P0.29,P2.9 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
```
After restart klippy all fine:
```
mcu 'mcu': Starting serial connect
Starting stk500v2 leave programmer sequence
Got b'' from stk500v2
new minimum rtt 929.248: hrtt=0.000378 freq=100000000
new minimum rtt 929.303: hrtt=0.000180 freq=100159355
new minimum rtt 929.411: hrtt=0.000167 freq=99960464
Loaded MCU 'mcu' 94 commands (v0.10.0-122-gc179db3d / gcc: (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] binutils: (2.34-4ubuntu1+13ubuntu1) 2.34)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c0=P0.28,P0.27 BUS_PINS_i2c1=P0.1,P0.0 BUS_PINS_i2c1a=P0.20,P0.19 BUS_PINS_i2c2=P0.11,P0.10 BUS_PINS_ssp0=P0.17,P0.18,P0.15 BUS_PINS_ssp1=P0.8,P0.9,P0.7 CLOCK_FREQ=100000000 MCU=lpc1768 PWM_MAX=255 RESERVE_PINS_USB=P0.30,P0.29,P2.9 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (1024 moves)

```

Sineos, 2021-12-18:
Please attach full and unmodified klippy.log, showing this behavior.

Vittaly, 2021-12-29:
After move klipoer from termux (android shell) the problem is not reproduced

closed 2021-12-29 by Vittaly