Klipper3d/klipper #5044: MCU shutdown and error

MatrixTechno, 2021-12-18:
Whenever I try to connect the pi to my printer, I get one of these two errors. I will restart the firmware around 20 times before it actually works, this happens every time and I'm not sure why. 
[image]
[image]

Sineos, 2021-12-19:
What is your host computer? Anything else running on it, like webcams etc?

MatrixTechno, 2021-12-19:
I'm using the original Pi model B Rev 2, I've since turned off my webcam and it seems to give me the error less, but I still get it. Is the original pi just not powerful enough? 
[image]

Sineos, 2021-12-20:
I cannot comment on Pi model B Rev 2 as I never used it with Klipper. I have one as part of my home automation and I know it is slower than my analog watch.
It is even slower than a Pi Zero and they are known to be easily overwhelmed by Klipper.

Sineos, 2022-01-07:
Seems the topic is concluded. Please do not hesitate to open a new issue on https://klipper.discourse.group if needed

closed 2022-01-07 by Sineos