Klipper3d/klipper #5046: Inaccurate endstop position when doing a delta calibrate

HCR-Mad-Bull, 2021-12-18:
Hi, I've got a delta printer and when doing a delta calibration the endstops positions are always out by +0.4 which makes the nozzle too low, it says that the height is 453.6 but in reality it should be 453.2. I can manually edit the output so it's correct, but I'm unsure if the other results are also inaccurate as well which makes me worried. 

Anycubic predator
Microswitch omron magnetic probe
Skr pro 1.2
Raspberry pi 3
Latest klipper updates as of 18/12
Fluidd

Any other info you need?

simonkuehling, 2021-12-23:
Hi @HCR-Mad-Bull, how exactly do you run the delta calibration - using the probe or manually ("paper test")?

HCR-Mad-Bull, 2021-12-26:
Hi @simonkuehling sorry about the late reply, I was using a probe.

simonkuehling, 2021-12-26:
Did you carefully calibrate your probe according to the [Probe Calibration](https://www.klipper3d.org/Probe_Calibrate.html) documentation?
Usually, due to mechanical flex in every system, probes trigger a bit below their actual z=0 plane, which would match your observation of the nozzle beeing too low.

closed 2022-02-07 by github-actions[bot]