Klipper3d/klipper #5048: action_respond_info does not seem to apply to loop in macro

rknall, 2021-12-19:
I have written a small test code in order to try out some wait macro idea. The code goes as follows:

```
[gcode_macro test_code]
gcode:
    {% set _test = params.TEST|default(0)|float %}

    {% if params.TEST is defined and _test > 0 %}
        M117 Testing ...
        {% for cnt in range(_test|int) %}
         G4 P{1000}
         {% if cnt > 0 and (cnt % 60) == 0 %}
          {action_respond_info("Waited %d seconds ( %d total )" % (cnt|int, _test|int))}
          M105
         {% endif %}
        {% endfor %}
    {% endif %}
```

Now, the idea would be, that I get a message every 60 seconds, accompanied with the output of M105. In reality, I get ALL messages immediately upon calling the macro, and the output of M105 every 60 seconds. That does not seem to be correct?

See 
```
15:13:50 AMB:27.1 /0.0 B:27.5 /0.0 C:26.2 /0.0 T0:26.2 /0.0
15:12:50 AMB:27.2 /0.0 B:27.6 /0.0 C:26.3 /0.0 T0:26.1 /0.0
15:11:50 AMB:27.1 /0.0 B:27.6 /0.0 C:26.3 /0.0 T0:26.1 /0.0 
15:10:50 AMB:27.2 /0.0 B:27.6 /0.0 C:26.4 /0.0 T0:26.1 /0.0
15:09:50 AMB:27.2 /0.0 B:27.6 /0.0 C:26.4 /0.0 T0:26.2 /0.0
15:08:51 Waited 300 seconds ( 360 total )
15:08:51 Waited 240 seconds ( 360 total )
15:08:51 Waited 180 seconds ( 360 total )
15:08:51 Waited 120 seconds ( 360 total )
15:08:51 Waited 60 seconds ( 360 total )
15:08:51 test_code TEST=360
```

I am running Klipper v0.10.0-184-gdd714fc7

rknall, 2021-12-19:
I am attaching the klipp.log as well, although nothing can be seen much in it anyhow, the printer is working in general
[klippy.log](https://github.com/Klipper3d/klipper/files/7740935/klippy.log)

Sineos, 2021-12-19:
See https://github.com/Klipper3d/klipper/issues/5023

closed 2021-12-19 by rknall