Klipper3d/klipper #5050: SET_VELOCITY_LIMIT causes toolhead to come to a stop

RealDeuce, 2021-12-20:
If `SET_VELOCITY_LIMIT ACCEL=x` is used instead of `M204 Sx` with the toolhead at max speed, the toolhead will come to a stop before the next move.

The following video is of the printer executing the following GCode:
```
G28
G1 Z20
SET_VELOCITY_LIMIT ACCEL=10000 ACCEL_TO_DECEL=5000
G1 X10 Y10 F10000
G1 X175 Y175 F30000
SET_VELOCITY_LIMIT ACCEL=5000 ACCEL_TO_DECEL=2500
G1 X340 Y340

SET_VELOCITY_LIMIT ACCEL=10000 ACCEL_TO_DECEL=5000
G1 X10 Y10 F10000
G1 X175 Y175 F30000
M204 S5000
G1 X340 Y340
M400
M112
```
https://user-images.githubusercontent.com/9758938/146727393-c75ed4d3-9fe7-4fb4-8eec-32a57aeee79f.mp4

If I comment out the `print_time = self.get_last_move_time()` line, this stops occurring.  It's not clear what this line is intended to do since print_time is not used in this method.

[klippy.log](https://github.com/Klipper3d/klipper/files/7743167/klippy.log)

wlhlm, 2021-12-20:
I have noticed similar behavior since SuperSlicer added settings for travel acceleration. The way that was implemented was by splitting travel moves in two and changing the acceleration in between to the acceleration value for the extrusion role following the travel move.

I was using an overwritten `M204` that used `SET_VELOCITY_LIMIT` to set `accel_to_decel` in addition to `accel`. This resulted in the toolhead slowing down in the middle of travel moves (as demonstrated in the video above). Using the Klipper-provided `M204` doesn't show this problem.

RealDeuce, 2021-12-20:
Yeah, commenting out the line mentioned in the issue should fix that.  The call to get_last_move_time() flushes the look-ahead, which appears to be what triggers the issue.

I was in the same boat, and started with the assumption that it was a SuperSlicer issue before I realized it was actually a Klipper issue and started poking around.  I didn't realize that M204 didn't show the behaviour until I was composing this issue since I use the following M204 macro:

```# Replaces M204 with one that also sets accel_to_decel
# Two main differences between this and the stock one:
# 1) If 'S' is included with 'P' and/or 'T', the lowest of the three
#    is used.  In stock, 'S' is always used if present.
# 2) If M204 does not include 'S', 'P', or 'T', or if the value is
#    not a valid float, accel is changed to the max_accel from the
#    printer.cfg file instead of throwing an error.
[gcode_macro M204]
description: Set Acceleration and Accel to Decel
rename_existing: K204
gcode:
    {% set def = printer.configfile.settings.printer.max_accel|float %}
    {% set accel = [def, params.S|default(def)|float, params.P|default(def)|float, params.T|default(def)|float]|min %}
    {% set accel_to_decel = accel * printer.configfile.settings.printer.max_accel_to_decel / def %}
    SET_VELOCITY_LIMIT ACCEL={accel} ACCEL_TO_DECEL={accel_to_decel}
```

I disabled that macro and the problem "went away", which was the final piece of the puzzle.

RealDeuce, 2021-12-31:
Fixed with the commit.

closed 2021-12-31 by RealDeuce