Klipper3d/klipper #5054: max_power is not available in output_pin section

Zodiak1993, 2021-12-21:
Hello, why is in not possible to configure max_power with an output_pin (PWM).

I want to limit a led panel to 80% because it is getting hot inside the enclosure.

hmortensen83, 2021-12-21:
Can you not just set the value to 0.8? Something like this:
[output_pin caselight]
pin: ?
value: 0
pwm: True

SET_PIN PIN=caselight VALUE=0.8

Zodiak1993, 2021-12-21:
Yes of course I could to this but it is controlled by a slider in mainsail. So I don’t want to be careful to only slide to 80 percent ;)

hmortensen83, 2021-12-21:
Maybe you can use scale with a value of 1.25
In the code it says:
value /= self.scale
So I guess value = 1 / 1.25 = 0.8

yzheka, 2021-12-22:
you can create gcode macro SET_PIN and replace existing SET_PIN_BASE where you check if pin is caselight then scale it down as @hmortensen83 written above

Sineos, 2022-01-07:
Seems the topic is concluded. Please do not hesitate to open a new issue on https://klipper.discourse.group if needed

closed 2022-01-07 by Sineos