Klipper3d/klipper #5064: Impossible to add options to main lcd menu (example for add manual mesh leveling)

neox3, 2021-12-23:
Hi, i am new to klipper but with the help of community, i have a mainsailos + klipper + klipperscreen (using kiauh) working in a kingroon kp3s (not printing).

the problem i find, is that i need to use manual mesh leveling to calibrate my bed, using the calibration from mainsail only moves the machine to one point and stops, nothing is show in the lcd or the web interface for continue calibration.

using google, documentation, etc i find that is needed to make a special menu for mesh level (also i finded that the menu working system, changed a time ago). 

All the example files that i find that uses the menu form :  [menu __main __prepare] gives error in mainsail after saving and rebooting.

The most recent info i finded here, is that in printer.cfg you have to add this:

**[bed_mesh]
speed: 100
horizontal_move_z: 5
mesh_min: 10,10
mesh_max: 170, 170
probe_count: 3,3
mesh_pps: 2,2
algorithm: lagrange
bicubic_tension: 0.2
move_check_distance: 5
split_delta_z: .025
fade_start: 1.0
fade_end: 10.0**

**[menu __main __setup __calib __meshBed __start]
type: command
name: Start probing
gcode:
G28
BED_MESH_CALIBRATE
TESTZ Z=-5**

**[menu __main __setup __calib __meshBed __zposition]
type: input
name: Z: {'%05.3f' % menu.input}
input: {printer.gcode_move.gcode_position.z}
input_step: 1
realtime: True**

**[menu __main __setup __calib __meshBed __zNudgeDown]
type: command
name: Nudge Down
gcode: TESTZ Z=-0.025**

**[menu __main __setup __calib __meshBed __zNudgeUp]
type: command
name: Nudge Up
gcode: TESTZ Z=0.025**

**[menu __main __setup __calib __meshBed __next]
type: command
name: Next
gcode:
ACCEPT
TESTZ Z=-5**

**[menu __main __setup __calib __meshBed __save]
type: command
name: Save & Restart
gcode: SAVE_CONFIG**

**[menu __main __setup __calib __bedScrews]
type: list
name: Adjust screws**

**[menu __main __setup __calib __bedScrews __start]
type: command
name: Start
gcode: G28
BED_SCREWS_ADJUST**

**[menu __main __setup __calib __bedScrews __accept]
type: command
name: Accept
gcode: ACCEPT**

**[menu __main __setup __calib __bedScrews __adjusted]
type: command
name: Adjusted
gcode: ADJUSTED**

**[menu __main __setup __calib __bedScrews __save]
type: command
name: Save config
gcode: SAVE_CONFIG**

**[menu __main __setup __calib __bedScrews __abort]
type: command
name: Abort
gcode: ABORT**

Ok, what is the problem? that i , after saving and rebooting, always have this error:
Section 'menu __main __setup __calib __meshbed' is not a valid config section

but I go over there, whatever I do, change what I change, I always receive the same error that makes it impossible for me to add anything to the lcd menu, which is essential to be able to use the manual mesh leveling.

Examples:  
[menu __main]  -> Section 'menu __main' is not a valid config section
[menu __main x]  -> Section 'menu __main x' is not a valid config section
[menu x]  -> Section 'menu x' is not a valid config section

Always the same! it is normal? how can i add a item to my lcd? i need to add the code that i pasted above.

the original documentation of klipper says that is as easy as:

**[menu some_name]
type:
   One of command, input, list, text:
      command - basic menu element with various script triggers
     input   - same like 'command' but has value changing capabilities.
                Press will start/stop edit mode.
       list    - it allows for menu items to be grouped together in a
                 scrollable list.  Add to the list by creating menu
                 configurations using "some_list" as a prefix - for
                 example: [menu some_list some_item_in_the_list]
       vsdlist - same as 'list' but will append files from virtual sdcard
                 (will be removed in the future)
name:
  Name of menu item - evaluated as a template.
enable:
  Template that evaluates to True or False.
index:
   Position where an item needs to be inserted in list. By default
   the item is added at the end.**

https://www.klipper3d.org/Config_Reference.html?h=menu#menu

but something is wrong, because what the documentation says, at least it doesn't work for me.

[image]

vladimir-poleh, 2021-12-26:
Please add following section to the config before section `[menu __main __setup __calib __meshBed __start]`

```
[menu __main __setup __calib __meshBed]
type: list
name: Mesh Bed
```

Sineos, 2022-01-07:
Seems the topic is concluded. Please do not hesitate to open a new issue on https://klipper.discourse.group if needed

closed 2022-01-07 by Sineos