Klipper3d/klipper #5067: python2 is deprecated

gratefulfrog, 2021-12-25:
I have been struggling with the Resonance Testing section of Klipper.
It seems that these instructions and much of the code were written for python2, but on rpi4, this is a non starter.
In particular:
numpy won't run:
$ libcblas.so.3: cannot open shared object file: No such file or directory
soltuion:
$ sudo apt-get install libatlas-base-dev
Then, the script:
~/klipper/scripts/calibrate_shaper.py
needs to be updated:
line 1:  #!/usr/bin/env python3        #### was python2 !!! 

It may be the same for the all scripts that specifically mention python2 ...

In the end, I got it to work, I think.

I really appreciate your work and am not criticizing, just trying to help.

Any questions, I am always available!
Bob

Sineos, 2021-12-26:
Thanks for your finding. 
It is a known topic and already addressed --> https://github.com/Klipper3d/klipper/pull/5011
Closing the issue here

closed 2021-12-26 by Sineos

gratefulfrog, 2021-12-26:
Cheers!
Great piece of software Klipper ! Well done!
Bob