Klipper3d/klipper #5071: Unit klipper.service not found

AlphaDragon601, 2021-12-26:
I'm trying to install klipper onto me ender 3 alongside octopi. Following the install process I get the the flashing portion where the guide states to run the comman `sudo service klipper stop` which returns an error of `unit klipper.service not loaded` so I do `sudo service klipper start` and get an error `unit klipper.service not found`. I've been told to run `systemctl start klipper` but that creates its own set of issues that make it seem like the wrong solution. I've yet to find any meaningful solution online, are there any other suggestions of what to try next? I've reflashed my pi 3 times now and tried following the install as close as possible but this issue is sustains.

Sineos, 2021-12-27:
Please post the output of the installation command `./klipper/scripts/install-octopi.sh`

AlphaDragon601, 2021-12-27:
> Please post the output of the installation command `./klipper/scripts/install-octopi.sh`

`###### Running apt-get update...
[sudo] password for pi:
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0                                         MB]
Reading package lists... Done
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed i                                        ts 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be ap                                        plied. See apt-secure(8) manpage for details.
`

Sineos, 2021-12-27:
Run
```
sudo  apt-get update --allow-releaseinfo-change 
sudo apt full-upgrade -y
```

then repeat the installation. This is due to a bug in Debian Bullseye

AlphaDragon601, 2021-12-27:
The update seems to have fixed all those issues so thank you very much! However now when I try to run a piece of test gcode I get a flood of the error ` mcu 'mcu': Unable to connect` cancelling the print. I believe the flash was correct and even redid it just to be sure. Klipper connects to the printer fine but any kind of input from the octoprint menu gives that same error.

Sineos, 2021-12-28:
Please post your klippy.log. What kind of board?

AlphaDragon601, 2021-12-28:
Stock ender 3 v1 board 8 bit
[klippyLog.txt](https://github.com/Klipper3d/klipper/files/7785989/klippyLog.txt)
this is all right on startup

Sineos, 2021-12-29:
Your [mcu] section does not look correct.
Make sure to follow the header information in https://github.com/Klipper3d/klipper/blob/master/config/printer-creality-ender3-2018.cfg to flash the board.
After this, use `ls /dev/serial/by-id/*`
It should report something similar to the following: `/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0`

Sineos, 2022-01-07:
Seems the topic is concluded. Please do not hesitate to open a new issue on https://klipper.discourse.group if needed

closed 2022-01-07 by Sineos