Klipper3d/klipper #5080: BLTouch clone not supported in Klipper. Always reports open or TRIGGERED despite trying every single combination

mortenmoulder, 2021-12-29:
I have read the contact page, and I have asked on the forum, on Reddit, on a few Facebook groups, and on a few Discord groups. I have come to the conclusion that Klipper has a bug, which is why I'm creating this issue.

I have a 5v BLTouch with the writing "3D Touch" on it. No logos or anything - just 3D Touch. In Marlin the probe works great. I can stow and deploy all day long, and the probe works great as a Z-STOP as well. However, no matter what I do in Klipper, I cannot get it to work. It always reports "probe: open" or "probe: TRIGGERED" no matter what I do.

To be clear: I have read the documentation pages ([1](https://www.klipper3d.org/BLTouch.html) and [2](https://www.klipper3d.org/Config_Reference.html)) multiple times. I have tried everything multiple times. I have heard from some people, that their BLTouch clone behaved the exact same way, and that they instead purchased a genuine BLTouch. This should not be a requirement to get Klipper to work, when it works fine in Marlin.

Okay, so now to the actual issue. I have tried wiring it to both Z-STOP and Z-PROBE on my SKR Mini E3 V2 board. I have tried every single pin variety such as:

```
sensor_pin: ^PC2
sensor_pin: ^!PC2
sensor_pin: PC2
sensor_pin: !PC2
sensor_pin: ~PC2
sensor_pin: ~!PC2
```

As well as for PC14 when connected to Z-PROBE on my board.

I have tried every combination of the touch modes:

```
probe_with_touch_mode: False
probe_with_touch_mode: True
```

And I have tried every combination with "triggered" in them:

```
pin_up_touch_mode_reports_triggered: True
pin_up_touch_mode_reports_triggered: False

pin_up_reports_not_triggered: True
pin_up_reports_not_triggered: False
```

I have NOT tried `set_output_mode` because I'm worried that it might break something. If not trying that was a mistake, let me know exactly what I should do, and if my SKR Mini E3 V2 supports it.

I have tried calling `BLTOUCH_DEBUG COMMAND=pin_down` followed by `QUERY_PROBE` and it said "probe: open". I then pushed it up using my nail (according to the documentation), and ran `QUERY_PROBE` again. It still said `probe: open`.

At this point I've probably tried over 50 different combinations on both Z-STOP and Z-PROBE, but nothing seem to be working.

I would like a developer to help me verify that either A) I did something wrong or B) Klipper has a bug (or hasn't implemented support).

I can also provide a copy of my Configuration.h and Configuration_adv.h files for Marlin 2.0.8.2 that works for my probe connected to Z-STOP.

Thank you.

KevinOConnor, 2021-12-30:
There have been a few reports of some recent clones that do not support "touch mode".  These probes will not succeed on the documented "touch mode test", but it may be possible that they still function as a probe.  To test this, go back to the initial config and proceed with the BLTouch "Z home and/or PROBE test" even though you were unable to complete the "touch mode test".  Be ready to pull power if the toolhead does not stop descending when you manually trigger the probe with your fingernail.

-Kevin

mortenmoulder, 2021-12-30:
> There have been a few reports of some recent clones that do not support "touch mode". These probes will not succeed on the documented "touch mode test", but it may be possible that they still [...]

Thanks for the reply.

So just to be clear: "Go back to the initial config" means all values **except** `sensor_pin`, `control_pin`, offsets, etc. should be removed? Remove `probe_with_touch_mode`, remove `pin_up_touch_mode_reports_triggered`, and so on from my config, then try homing and touch the probe with my nail once it starts to home the Y axis?

KevinOConnor, 2021-12-30:
Alas, I can't really provide meaningful guidance on this - I've never owned a bltouch nor clone.  I've seen reports that the "touch mode test" doesn't work but that otherwise the probe does.  If you're working on this and can confirm that is the case then let me know.  We can then update the docs with a note on procedure for some clones.

-Kevin

grantemsley, 2021-12-30:
I can help with this since I have a similar clone and wasted a couple hours on it. I also commented in #4729

Configure it exactly as described in the documentation. You need at least the sensor and control pins set as specified in the docs.  `probe_with_touch_mode` must be False or not specified (touch mode doesn't work on these clones). `pin_up_reports_not_triggered` and `pin_up_touch_mode_reports_triggered` should also be left unspecified or set to their default of True.

Mine looks like this:
```
[bltouch]
# DO NOT enable touch mode, doesn't work on this clone probe!
sensor_pin: ^PA2
control_pin: PA1
x_offset: -38
y_offset: -9
samples: 2
speed: 1
sample_retract_dist: 2.0
lift_speed: 1
samples_result: average
samples_tolerance: 0.100
samples_tolerance_retries: 0
```

The problem comes in when you try to follow the docs for testing it. ``BLTOUCH_DEBUG COMMAND=pin_down`` and ``BLTOUCH_DEBUG COMMAND=pin_up`` should work. If not you have wiring/config issues. But ``BLTOUCH_DEBUG COMMAND=touch_mode`` will not trigger the probe. Touch mode doesn't work.

So skip over testing touch_mode, make sure the bed and nozzle are very far apart, and try homing the machine (I'm assuming you are using ``endstop_pin: probe:z_virtual_endstop`` for your Z stepper. Make sure it pops out, and trigger it with your finger. If the Z axis doesn't stop moving, hit emergency stop. If it does stop moving, then the probe is working properly. Homing will fail because it has to trigger twice in roughly the same place, but reset the machine and try homing again against the bed - be ready to hit emergency stop if something goes wrong. If homing works, then you should be good.

The clone I have has about the same repeatability as my other official one as long as I'm moving the bed slowly. The only real downside is you can't use touch mode where it probes the bed without retracting and extending for each probe, so it takes a little longer.

mortenmoulder, 2021-12-30:
@grantemsley  Huuuuuuuuuuuuuuuuuuh that's SO weird. I've been relying on `QUERY_PROBE` all this time (except a few times.. like 5 times), and if I use your configuration as well as `z_offset`, it actually stops. That's so weird. How the heck? So is `QUERY_PROBE` bugged?

So freaking weird. But yes, `G28 Z` works flawlessly. It can perfectly home now without any issues what so ever. WEIRD. THANK you so much for helping out with that piece of information.

KevinOConnor, 2022-01-04:
PR #5088 should address this.

-Kevin

closed 2022-01-11 by github-actions[bot]