Klipper3d/klipper #5081: Temperature reading

tony21675, 2021-12-29:
I recently upgraded my hot end to a E3D-V6. Once I put it on and updated klipper and mainsail to the newest version both my hot end and bed quit sending accurate temperatures. Both are reading around 200c with no heat being applied. I have tried 3 different thermistors thinking one could be bad with no luck.

[klippy-2.log](https://github.com/Klipper3d/klipper/files/7790752/klippy-2.log)

Sineos, 2021-12-30:
Most likely a wiring issue. Loose connector, broken cable, short circuit

tony21675, 2021-12-30:
Later today when I get a chance I will take it all apart and put it together peace by peace and see where it might be wrong.

tony21675, 2021-12-30:
Ok I disconnected both my hot end and bed (thermistor and power leads) still reading 180c plus. So I think my board is blown or something is wrong in my cfg file.

# This file contains common pin mappings for the BIGTREETECH SKR V1.4
# board. To use this config, the firmware should be compiled for the
# LPC1768 or LPC1769(Turbo).

# See docs/Config_Reference.md for a description of parameters.

[include mainsail.cfg]

[stepper_x]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
microsteps: 32
rotation_distance: 31
full_steps_per_rotation: 200
endstop_pin: tmc2209_stepper_x:virtual_endstop
homing_retract_dist: 0
position_endstop: -12
position_min: -12
position_max: 200
homing_speed: 40

[tmc2209 stepper_x]
uart_pin: P1.10
diag_pin: P1.29
driver_SGTHRS: 95
run_current: 0.400
stealthchop_threshold: 999999

[stepper_y]
step_pin: P0.19
dir_pin: P0.20
enable_pin: !P2.8
microsteps: 32
rotation_distance: 31
full_steps_per_rotation: 200
endstop_pin: tmc2209_stepper_y:virtual_endstop
homing_retract_dist: 0
position_endstop: -41
position_min: -41
position_max: 200
homing_speed: 40

[tmc2209 stepper_y]
uart_pin: P1.9
diag_pin: P1.28
driver_SGTHRS: 135
run_current: 0.400
stealthchop_threshold: 999999

[stepper_z]
step_pin: P0.22
dir_pin: P2.11
enable_pin: !P0.21
microsteps: 16
rotation_distance: 8
full_steps_per_rotation: 200
endstop_pin: probe:z_virtual_endstop
position_min = -3
position_max: 300
homing_speed: 50

[tmc2209 stepper_z]
uart_pin: P1.8
diag_pin: P1.27
run_current: 0.650
hold_current: 0.450
stealthchop_threshold: 999999
[... 244 lines cut ...]
[output_pin beeper]
pin: P1.30

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  -0.265000, -0.105000, -0.062500, -0.082500, -0.215000
#*# 	  -0.192500, -0.050000, 0.000000, -0.047500, -0.202500
#*# 	  -0.162500, -0.057500, -0.050000, -0.102500, -0.245000
#*# tension = 0.8
#*# min_x = 25.0
#*# algo = bicubic
#*# y_count = 3
#*# mesh_y_pps = 2
#*# min_y = 25.0
#*# x_count = 5
#*# max_y = 185.0
#*# mesh_x_pps = 2
#*# max_x = 185.0
#*#
#*# [bltouch]
#*# z_offset = 2.000

Sineos, 2021-12-30:
BTT SKR1.4? If yes, you have a third temp port with pin P0.23

tony21675, 2021-12-30:
Yes it is giving false reading on both bed and hot end. Port 3 might be ok but port 1 and 2 both seem to be malfunctioning

tony21675, 2021-12-30:
BTT SKR1.4 correct

Sineos, 2022-01-07:
If the issue still persists, please do not hesitate to open a new topic on https://klipper.discourse.group

closed 2022-01-07 by Sineos