Klipper3d/klipper #5087: Does probe:z_virtual_endstop respect the probe's x and y offset? If not: should it?

jue89, 2021-12-30:
I'm using a BLTouch as a virtual endstop. I've noticed that `[safe_z_home]` moves the nozzle to the specified `home_xy_position`. But shouldn't the probe moved to that position instead?

I've tried to find the answer in the code (I have to admit that I'm a Python noob) but it looks like the offset is just used for the bed mesh leveling. The virtual endstop assumes that the bed has the same z offset for nozzle and probe at the homing position.

Am I correct with the assumption that the probe should be moved to that position?

Sineos, 2021-12-30:
Movement is relative to your head and not the probe

jue89, 2021-12-30:
Yes, right. But the endstop should detect when the nozzle touches the surface, right? But the probe, that is capable of detecting this event, is located at the specified offset and not at the position of the nozzle.

For example: 
- Let's have a probe at offset position: `x_offset = 40`, `y_offset = 0`, `z_offset = 0`
- The bed is rotated by an angle of 1° around the y-axis
- Due to the rotation the probe will trigger `40mm x tan(1°) = 0.7mm` too early

If the probe is moved to the `home_xy_position` instead of the nozzle, that error wouldn't occur.

Or have I gotten something wrong?

Sineos, 2021-12-30:
This is where your bed mesh comes into play. It generates a height map of your bed and the correction is of course applied to the tool-head, taking the x / y offsets into account.
Apart from this the applicable z_offset should be determined via https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-z-offset which also takes into account the offset.

Regarding your example, a tilted bed will anyway create troubles and there are functions in place to level your bed, e.g. https://www.klipper3d.org/Manual_Level.html?h=screw#adjusting-bed-leveling-screws

jue89, 2021-12-30:
> This is where your bed mesh comes into play. It generates a height map of your bed and the correction is of course applied to the tool-head, taking the x / y offsets into account.

Okay, I see this issue is an academic one for those making use of bed mesh levelling. Users must recalibrate if the home position is moved, right?

> Regarding your example, a tilted bed will anyway create troubles and there are functions in place to level your bed, e.g. https://www.klipper3d.org/Manual_Level.html?h=screw#adjusting-bed-leveling-screws

Sure. I just needed an example that show cases my thoughts.

Sineos, 2021-12-30:
> Users must recalibrate if the home position is moved, right?

I'd say this depends on your bed. If it is badly warped or has constantly shifting screws. My approach is to chose the location of a screw as reference point for calibrating the z-offset and keep this point also as reference for adjusting the screws. 
At least to my experience, I cannot complain about first layer stability

jue89, 2021-12-30:
Okay. Thank you for your replies!

I'll guess there's nothing to be fixed here. All possibly made errors are compensated with bed mesh levelling.

closed 2021-12-30 by jue89