Klipper3d/klipper #5089: RepRap Display doesn't work properly

Stout69, 2021-12-31:
Hi folks 👋 
Here is situation. My Klipper works on Raspberry Pi4 + BTT SKR2 (rev.b). Everything seems looking good but I have an issue. The display which is named as "RepRapDiscount 128x64 Full Graphic Smart Controller" (st7920) shows picture incorrectly with some line shifts which makes "broken picture" feelings. 

[image]

Pins for that display configured as an example in `sample-lcd.cfg`:
```
##########################
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC5, EXP1_3=PB1, EXP1_5=PE10, EXP1_7=PE12, EXP1_9=<GND>,
    EXP1_2=PB0, EXP1_4=PE9, EXP1_6=PE11, EXP1_8=PE13, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PA6, EXP2_3=PE7, EXP2_5=PB2, EXP2_7=PC4,   EXP2_9=<GND>,
    EXP2_2=PA5, EXP2_4=PA4, EXP2_6=PA7, EXP2_8=<RST>, EXP2_10=<NC>

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8
##########################
```
At the sametime the same display on the same motherboard connected with the same cable but on Marlin firmware works perfectly well! No any line shifts. 🤔 

Does anyone have any ideas how to make display works with Klipper?

I tried to edit the following file `/home/pi/klipper/klippy/extras/display/st7920.py` and increased (as I thought) delays  
```
ST7920_CMD_DELAY = .000200
ST7920_SYNC_DELAY = .000200,
```
deleted _st7920.pyc_, did `make clean`, compiled firmware, deployed to SKR2, but nothing changed.

AlexRoman, 2022-01-03:
I can confirm that I am seeing the same issue with the RepRap display. Working fine on Marlin, but not with Klipper. I've also tried playing with the timing parameters to no avail.

Stout69, 2022-01-30:
### Problem Solved!
**WARNING! Solution was tested ONLY on Raspberry Pi4 + BTT SKR2 configuration. How that will work on another configuration and hardware I don’t know unfortunately.**
_Brief overview (For whom tries to avoid technical details)._
If you have the following display
[image]
and the following issues, it (potentially) can help you to fix them by replacing these two files:
/src/lcd_st7920.c and /klippy/extras/display/st7920.py
Updated files can be found here: [(https://github.com/Stout69/For-Klipper)](https://github.com/Stout69/For-Klipper)
The 1st file is a part of MCU software, that means when you replace that file (by updated copy) you should recompile MCU firmware, i.e. (on Linux) run commands make clean, make (or make flash) and if your motherboard can not be refreshed directly from the host,  write updated MCU firmware  in appropriate way.
_The story in details._
To understand the reasons for the “incorrect” functioning of LCD I connected the following two displays in parallel – the 1st buggy «RepRap» display and the 2nd Ender3 display (which works perfectly without any issues with the default Klipper software). Both displays work under ST7920 protocol with Marlin firmware with no issues. I’ve connected a logical analyzer to the following pins CS, SCLK, SID and started observing.
For the beginning I realized the difference between Klipper and Marlin, especially how they draw information on display.  
Marlin manages the CS line with every package, while Klipper sets CS on logical 1 at the beginning of work and does nothing with it after that.
Marlin has a high Idle state for SCLK line, while Klipper has low.
And finally, more importantly, Marlin outputs data each second, refreshes the display in graphical mode (writes data into GDRAM) and doesn’t use an internal character generator (which is located in display hardware). Klipper outputs data only if content on display is changed and specially outputs parts which are changed. Klipper in general uses text mode, when symbols codes are written in DDRAM, but appearance of the symbols are taken from the internal character generator of the display hardware.
After that I wrote a few test programs and found out that clone of st7920 which are installed into «RepRap» display works incorrectly when outputting symbols from their internal character generator. For example, program (on picture) outputs 4 characters in sequence:
[image]
As we can see, if the output does not start at the beginning of the line  (the address is not a multiple of eight) then there is a vertical shift according to the formula n * 8 pixels (0 < n < 4). So, if you output text in lines of 16 characters, starting from the first or second character space of the line, then there are no problems, but if from the third or more, the character will be shifted vertically.
Moreover, this chip, unlike the one installed in Ender3, does not combine pixel-by-pixel output from graphics memory and DDRAM. The latest data that came in completely replaces what came before it, and is not combined by XOR. 
To solve that issue we need to replace output into DDRAM with output into GDRAM. I’m imaging 3 options: 

- Install middleware microcontroller between the display and Klipper MCU
- Change Kipper software on host.
- Change Klipper software on MCU

The first option I declined because of money costs.
I program only on C and VB, but the host part of Klipper is written on Python (and honestly at that time I knew about Python only that it exists) and from two other options I preferred the 3rd one. 
I rewrote some software parts in file src/lcd_st7920.c . I wrote there a command interceptor from the host, three buffers (graphical, text and user’s symbols) & etc. When the appropriate commands arrived from the host, graphic data was written to its buffer, text data was converted to graphical form and written to its buffer, at the output the data of these buffers were combined by XOR and transferred to the display in graphic mode. The display worked great! Only now, when the contents of the screen changed (for example, when the table heating was turned on), the connection between the host and the MCU was interrupted and the system went into emergency stop mode. To be honest, it really surprised me. I did not expect that this would have such a serious load on ARM 168 MHz. But I had to refuse such a decision and I went to get acquainted with Python))) 
Finally I changed the following:
Into initializer of  DisplayBase class I’ve added to buffers:

   `self.CacheTextAsGraph = [bytearray(32) for i in range(32)]`
   `self.CacheGraphics = [bytearray(32) for i in range(32)]`

I’ve added method upd_graph:

```
 def upd_graph(self, x, y, lenX, lenY):
    for i in range(y,y+lenY):
         for c in range(x,x+lenX):
                self.graphics_framebuffers[i][c] =
                    self.CacheTextAsGraph[i][c] ^ self.CacheGraphics[i][c]
```

I’ve changed methods write_text and write_graphics:

```
def write_text(self, x, y, data):
            if x + len(data) > 16:
              	  data = data[:16 - min(x, 16)]
            gfx_fb = 16 * y
            if gfx_fb >= 32:
              	  gfx_fb -= 32
              	  x += 16
```
          
            for i, asc in enumerate(data):
              	  if type(asc) is str:
                    c=ord(asc)
              	  else:
                    c = asc
              	  if c > 6:
                	       for r in range(16):
[... cut ...]
After that changes the display worked perfectly. At first I left the original MCU firmware, but over time I found out that if SCLK IdleState=0, the display starts “once”, and if the CS line is not controlled, then the display sometimes freezes during prolonged operation. Therefore, I changed the /src/lcd_st7920.c file, specially the st7920 structure (added the cs entity there) and slightly rewrote the st7920_xmit_byte, command_config_st7920, command_st7920_send_cmds and command_st7920_send_data procedures. 

Side Note: Duck typing - that may be a good thing for language flexibility, but it’s a nightmare for reading and analyzing source code for somebody else.

ros3dprinter, 2022-02-02:
Hi, 
I have a similar problem. How do you applied changed files? Can write two lines on how to do it?
I located the two files. I suppose need to be changed with ones you provided.
After that? just a "make" on klipper root directory? or some other way?

I don't know if someone already checked the issure and applied changes on latest version of klipper (I think no due the fact your proposal is of only 12 days ago).

Regards

Stout69, 2022-02-05:
Hi,
You need to replace the original files which are located in Klipper (inside the Raspberry Pi) with the following fies i.e. st7920.py file need to be copied into /klippy/extras/display/ directory (which is also located in inside the Raspberry Pi), and  lcd_st7920.c file in the /src/ directory and replacing the files lying there. I do it with WinCSP. After that, you need to recompile the firmware of the MCU board in accordance with the instructions for your board. (Usually this is done with the "make" command from under Putty). With the resulting file (klipper.bin) you need to flash your slaveboard (as you did during the initial installation of Klipper) and reboot Klipper.
This solution will work as long as Kevin does not change the structure of the project sources, i.e. if in new versions lcd_st7920.c and st7920.py are in the same places as now, then the solution will work. It is possible that after each update it will be necessary to overwrite these files again, or maybe not, I have not checked.

Regards

closed 2022-03-12 by Stout69

ros3dprinter, 2022-03-13:
Hi,

Sorry for missing answer.

Yes, issue was resolved. You can close ticket.

Really thanks for your support

Regards

Rosario

yoerivdm, 2022-07-31:
Just noticed the same issue ... isn't it an option to add a 'new' type of display in Klipper without the need of fiddling with the files ourselves? For me it's no problem to do so but non-technical users will be left in the cold ...