Klipper3d/klipper #5093: MCU 'mcu' shutdown: Timer too close

HelgeKeck, 2022-01-01:
Hi,

im getting the famous MCU 'mcu' shutdown: Timer too close error, when using a additional MCU.

Running the latest Klipper on a Raspberry Pi4 4GB with an Octopus V1.1 Board.

The error happens randomly when i try to use some manual steppers from my MMU2S unit. I thought its because im using an old Trigorilla board for it, but after switching to a brand new SKR Mini V2.1 the error still happens. 

The error happens once in a week or so, the last time (the attached klippy log), it happend after the print was already finished and the MMU tried to eject the filament. 

As far as i can tell, it happens always when i try to sync a manual stepper with the extruder stepper, like:

```
MANUAL_STEPPER STEPPER=gear_stepper MOVE=-20 SPEED=10 SET_POSITION=0 SYNC=0
G1 E-20 F600
MANUAL_STEPPER STEPPER=gear_stepper SYNC=1
```

The raspberry pi defintiely didnt do anything unusual in this moment, so i dont know how to fix that

thanks

PS: not sure if this is correct, but i had to zip the klippy log in order to upload it here.

[klippy.log.zip](https://github.com/Klipper3d/klipper/files/7797476/klippy.log.zip)

HelgeKeck, 2022-01-01:
so it just happed again and i thought i share the new logfile

[klippy.log.zip](https://github.com/Klipper3d/klipper/files/7797768/klippy.log.zip)

namelessFPV, 2022-01-17:
[klippy (1).log](https://github.com/Klipper3d/klipper/files/7882564/klippy.1.log)

Teila, 2022-01-26:
I am also receiving this issue with my ERCF mcu (seeduino). This has also happened outside of printing via GCode IE just testing tool changes
[klippy (11).log](https://github.com/Klipper3d/klipper/files/7946381/klippy.11.log)
.

namelessFPV, 2022-02-16:
I fixed it whit OC.
I overclocked my pi zero 2W to 1.1Ghz or something like that.
And locked it to that.

jrhelbert, 2022-02-16:
I've been fighting this for a couple months now. Raspberry Pi 4 with a BTT Octopus. Mainsail/moonraker/klipper are the only things running on it. I have a webcam running, but have seen the error both with and without the webcamd service disabled.

It always happens in the first part of my print, while printing the skirt. I can always tell its about to happen because my extruder goes into "funky noise land" and I get some really crazy movements and overextrusions. You can see in my klippy.log that it actually triggers my filament encoder sensor and if I resume the print and let it go I will end up with the aforementioned mcu error.

If I do a firmware restart and reprint the exact same gcode it will have no problems and print successfully.

My utilization numbers on the pi are usually only around a 0.2 load average and at 94% idle. I've done everything I can think of, disabling power saving the pi so it never clocks down, disabled as many services as I can think of, decreasing my microsteps on all steppers, etc. Nothing has seemed to improve the issue

[klippy.log_mcu_error_011222.log](https://github.com/Klipper3d/klipper/files/8080760/klippy.log_mcu_error_011222.log)
.

jrhelbert, 2022-02-16:
> not sure if it is important, im using TMC 2209s in UART Mode

I am as well.

hugoalves6, 2022-02-27:
Having this issue also, im using pi zero 2w, with fysetc spider with all tmc2209, trying to do my own color selector, but when i try to move the manual_stepper less than 10-20sec of each movement it triggers this kind of error...

SkyLuke91, 2022-03-21:
Hi there I'm having same issue.
The error occurs randomly, motly when printing longer prints. (For example i had no issue printing 2h prints, but when i try to print 8h print it stops after 1hour or so [randomly])
This one failed in layer #13
[HT50 HOLDER-PLA-0.2mm-10%-3h10m-55g.txt](https://github.com/Klipper3d/klipper/files/8314033/HT50.HOLDER-PLA-0.2mm-10.-3h10m-55g.txt)
Running on Rpi zero 2W, Moontracker, Fluidd, Klipper without any cameras.
Gcode from SuperSlicer without any postprocess.
Klippy log attached.
[klippy.log](https://github.com/Klipper3d/klipper/files/8314013/klippy.log)

Is there something I can do to eliminate those?
Thanks

Sineos, 2022-03-21:
This topic is getting out of control.
The first few relates to `MANUAL_STEPPER` and the rest is mainly the "generic" `Timer too close` issue, probably mostly caused by weak or overloaded Pis.

If the SYNC topic is still present, it would make sense to open a new thread on https://klipper.discourse.group and likewise for the unrelated to SYNC, please open an own topic or search on discourse as enough topics are already existing.

psyvision, 2022-03-21:
Another user with `Timer too close` errors when using `MANUAL_STEPPER` even with `SYNC=1`.  I am running multiple MCUs, TMC2208 driving the manual stepper.

[klippy (10).log](https://github.com/Klipper3d/klipper/files/8315673/klippy.10.log)

closed 2022-03-23 by github-actions[bot]

iplayfast, 2022-07-24:
I've just updated my software (klipper/moonraker/mainsail on raspberry pi3, octopu mcu).

Navyguy330, 2022-08-22:
> This topic is getting out of control. The first few relates to `MANUAL_STEPPER` and the rest is mainly the "generic" `Timer too close` issue, probably mostly caused by weak or overloaded Pis.
>
> If [...]

friend is having this same issue with a 8gb pi.  Hardly an overloaded pi.  klipper don't need jack shit in terms of resources. a pi zero would run this.  

Its something else and every thing I have seen on google means not ONE person actually knows jack shit about the error or even any where close to how to fix it.

phil2sat, 2022-10-07:
Don`t know if its the issue, but on my Pi zero W i got the same. Dirven me Mad, sometimes 20 times reconnect until it got connected. What it fixed for me was:

[delayed_gcode delayed_homing]
gcode:
    G28
    STAR_TREK
initial_duration: 8  <--- Set this from 0.1 to a value higher wich works

So on connection its time to settle down before the Printer begins to auto home and so on.

jpsilvashy, 2022-11-10:
It's a shame this issue is getting so little attention and this issue was closed. I too have a Pi4 B+ with a 16gb v90 UHS-II SD card, no webcam, no other software running on the pi besides klipper and mainsail, both updated to the most recent verson.

Sineos, 2022-11-10:
Sorry, but it is the right thing:
1. The OPs issue was very specific
2. All the following me-too's had the same error message but not the same setup
3. The error can come from  multiple sources: Unstable voltage, errors on the SD, other devices hogging the USB bus, out of memory, CPU load, too much IO activity, overheating, other throttling scenarios etc
4. GH is not used to work on individual error scenarios
5. Open an issue on https://klipper.discourse.group with your individual case. Make sure you have checked / ruled out all under 3. because it is the first answer you will receive there as well