Klipper3d/klipper #5099: CR3D probe reading wrong values sometimes

ph3x, 2022-01-02:
Greetings,
for no obvious reasons I´m getting wrong values from my CR3D probe one some prints.
I´m slicing with SuperSlicer and using built in function to upload the gcode to the printer.
Today I had several successful prints and the last one was again with wrong values.
The offset is about ~0,2 larger so nozzle is hitting the plate. 

Klipper version v0.10.0-220-g608cd38d

Logfile and config attached.

19:28:39
probe at 0.000,-4.000 is z=0.492500
^print OK!
20:50:43
probe at 0.000,-4.000 is z=0.702500
^print NOK!

If there is any more information I can provide just tell me.

Cheers,
Christian

[bugreport.tar.gz](https://github.com/Klipper3d/klipper/files/7799941/bugreport.tar.gz)

OXERY, 2022-01-02:
I also know a few guys having these problems when starting another print without switching off the printer or restarting Klipper. The probed Z-Offset seems to increase after each print, like you mentioned. The nozzle scratches the bed and damages it.
The printers affected seem also to be Prusa or similar - in every case a Pinda or CR3D magnetic probe is attached.

simonkuehling, 2022-01-05:
Could it be a temperature drift in the magnetic sensor? These sensors are usually affected by temperature in a way that their offset changes quite a bit. (compare https://github.com/prusa3d/Prusa-Firmware/issues/16)

I would check to see if you can optimize your start gcode so that the probe will stay far away from the print bed while heating up.

StefanMeGit, 2022-01-18:
I rescently got the same probe and dont have any issues. Works since 5 days without any failure. Strange... what buildplate are u using?

ph3x, 2022-01-18:
I'm using spring steel with pertinax plate on top. Also I don't think it's
a temp issue since it also happened while printing same parts over and over
again with same workflow and also same nozzle position.

closed 2022-03-02 by github-actions[bot]