Klipper3d/klipper #5100: bltouch: nozzle crash during bed_mesh_calibrate, probe_with_touch_mode=False

harikattar, 2022-01-03:
Authentic BLTouch, klipper version 608cd38de (dec 31, 2021)
Narrowed it down to the interaction of two config lines:
[bltouch]
stow_on_each_sample=False
probe_with_touch_mode unset (default=false?)

Setting either to True fixes it.

When doing a home XYZ, the bltouch pin deploys correctly.  Following that with BED_MESH_CALIBRATE it moves to the first position, does NOT deploy the pin, and crashes the nozzle into the bed. 

Not sure exactly when this broke, I've updated a system from febuary 2020 to current head, so that's a rather large range to narrow down.  bed mesh with bltouch worked previous to the update.

adding probe_with_touch_mode=True to the config made it work as expected.

I think the easiest fix would be to make this configuration a fatal error since it results in potential damage.  It's already recommended against in the documents.  I only encountered it because probe_with_touch_mode was added after(?) the last time I did an update so it wasn't in my config file.

[bed_crash.log](https://github.com/Klipper3d/klipper/files/7800188/bed_crash.log)

dmerillat, 2022-01-03:
Upon further testing, this may be a race condition because the same configuration that was working is now not working.  If I don't find anything obvious I'll work on bisecting.  It's also possible the head crash damaged something so I'll re-test with the old firmware and configuration.

harikattar, 2022-01-03:
upon further testing this turned out to be an incredibly coincidental loose wire that worked/failed in tune with configuration changes through 4 iterations of doublechecking my settings.  Heisenbugs are the best bugs.

closed 2022-01-03 by harikattar