Klipper3d/klipper #5103: cartesian dual z motor with belts won't swap direction #5102

stevetomassetti, 2022-01-03:
hello 
ive go this ender 3v2 converted to dual z belts. I've installed bigtreetech octopus v1.1 with TMC 2209s .  I've  tried swapping rotation by placing ! on the DIR pins for z, z1 ( I've tried all combinations possible ) nothing seam to swap rotations on z or z1. if some one could help me out that would be greatly applicated
[klippy (4).log](https://github.com/Klipper3d/klipper/files/7803633/klippy.4.log)

Sineos, 2022-01-04:
If your steppers are moving into the wrong direction then invert the `dir_pin`, e.g.  `dir_pin: !xyz` instead of `dir_pin: xyz`

Your log shows a lot of TMC related errors. You should check your drivers and cables and board jumper settings. Also see https://www.klipper3d.org/TMC_Drivers.html#querying-and-diagnosing-driver-settings

schnello, 2022-01-10:
Please note that Port 2 and 3 are linked.
[image]

closed 2022-02-22 by github-actions[bot]