Klipper3d/klipper #5105: Klipper can't connect to printer

Viktor2O2, 2022-01-04:
Printer: creality ender 3v2

I'm trying to install klipper, but I'm having issues getting it to connect to my printer.
I follow the guide on how to install klipper.

In the klipper.log file this is the error (mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/serial/by-id/<your-mcu-id>: [Errno 2] No such file or directory: '/dev/serial/by-id/<your-mcu-id>') which is weird because in my printer.cfg file this is what is says.
[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

So I thought klipper may be using the wrong printer.cfg file, but as far as I know this is not the case.
I checked in the logs where the printer.cfg file was located and after downloading the file through FileZilla it turned out it was the same.

I have also tried reflashing the firmware multible times, but nothing seems to work.

[klippy.log](https://github.com/Klipper3d/klipper/files/7807613/klippy.log)
[printer.txt](https://github.com/Klipper3d/klipper/files/7807618/printer.txt)

Thanks in advance
Viktor

Sineos, 2022-01-04:
Please closely follow the information on top of https://github.com/Klipper3d/klipper/blob/master/config/printer-creality-ender3-v2-2020.cfg and https://www.klipper3d.org/Installation.html

The mcu serial is individual to your printer and needs to be obtained as shown in the above mentioned documentation. It will only be available when you have correctly flashed your board

Viktor2O2, 2022-01-04:
Thank you for the reply,

I read through the steps again, and I'm pretty sure that's exactly what I did.
It is getting late where I live, so I will check to make sure tomorrow.

I would just like to mention that I am not running klipper on a raspberry pi I don't know if this could be causing any issues.
I'm using an old mac mini that I use as an ubuntu server. (I can't find any store that has a raspberry pi in stock).

Viktor2O2, 2022-01-05:
Okay so to make sure I did nothing wrong I completely reinstalled klipper, moonraker and fluid.
After reflashing the printer and following all the steps again it finally worked!
I don't know what I did wrong, maybe it could have something to do with the fact that I originaly deleted the printer.cfg file and installed a new one (I have no idea).

Anyway thanks for your support and I'm sorry for wasting your time

closed 2022-01-05 by Viktor2O2