Klipper3d/klipper #5106: FR: Add Parameter WAIT to SET_HEATER_TEMPERATURE

Markus-1984, 2022-01-04:
Hi,

in order to get the status LEDs of my VORON Stealtburner Printhead on when the printer is heating up, i have to customize the Heatup Gcode. Unfortuanly i can´t to this with the M109 or M190 so i have to do it with the SET_HEATER_TEMPERATURE makro like this:

[gcode_macro SET_HEATER_TEMPERATURE]
rename_existing:             _SET_HEATER_TEMPERATURE 
gcode:
    STATUS_HEATING
    _SET_HEATER_TEMPERATURE {% for p in params
      %}{'%s=%s ' % (p, params[p])}{%
      endfor %}
    {% if params.WAIT %}
    TEMPERATURE_WAIT SENSOR={params.HEATER} MINIMUM={params.TARGET}
    {% endif %}

I can now call SET_HEATER_TEMPERATURE HEATER=extruder Target=250 WAIT=True for example and it lights up the LEDs, heats the hotend up and wait.

It works well but would be nice if the WAIT parameter could be implemented to the SET_HEATER_TEMPERATURE Gcode

simonkuehling, 2022-01-04:
Hi @Markus-1984 - why can't you do the same with M109/M190? You should be able to choose any "traditional gcode style" parameter like "W" for example - as in `M109 S70 W1` - to do the same with these commands.

To make your code a little bit cleaner, I guess you could use the [rawparams](https://www.klipper3d.org/Command_Templates.html#the-rawparams-variable) variable to pass the params through to _SET_HEATER_TEMPERATURE. Other than that, your solution looks like a very good one.

Markus-1984, 2022-01-04:
Hi @simonkuehling,

thanks for your tip with rawparams. I´ve just copy and paste the way I do from the klicky macros ;-)
But with rawparams it looks way easier.

The reason why I can´t do the same with M104 or any other classical Gcode command is, it´s not supported.
See here: https://github.com/Klipper3d/klipper/issues/335#issuecomment-589968789

simonkuehling, 2022-01-04:
that's right, you cannot rename a "traditional style" command into an "extended style" command (eg, from M109 to SET_EXTRUDER). But you should be able to use it like this (untested):

```
[gcode_macro M109]
rename_existing: M109.1
gcode:
    STATUS_HEATING #Set the Stealthburner LEDs
    M109.1 { rawparams }

[gcode_macro M190]
rename_existing: M190.1
gcode:
    STATUS_HEATING #Set the Stealthburner LEDs
    M190.1 { rawparams }
```
Is this more like what you are aiming at?

Markus-1984, 2022-01-04:
I´ve only tried this:

[gcode_macro M109]
rename_existing: _M109
gcode:
    STATUS_HEATING #Set the Stealthburner LEDs
    _M109 { rawparams }

Therefore I got the "G-Code macro rename of different types" error message.
I don´t mentioned the .1 method :-D

simonkuehling, 2022-01-04:
You can also choose any other name that follows the traditional syntax of `[single character][number][.(optional)][number(optional)]` (if this makes sense). So for example rename it to M1000 and M1001 or such. Does not have to be a ".X", but I like that because the command is still recognizable afterwards...

Markus-1984, 2022-01-04:
Thanks for your advise

Markus-1984, 2022-01-04:
Got another question about gcode_override.
Is it possible to do multiple Gcode overrides?
For example in the klicky config are many overrides like [z_tilt_adjust] or [bed_mesh_calibrate].
In order to get the SB LEDs working when the gcodes are called, i´ve added the macro call for the status into the klicky config file.
Would it be possible to make a seperate SB config file and do the gcode override in there?
It would be a double gcode override.

simonkuehling, 2022-01-04:
You cannot have multiple of the same `[gcode_macro M104]` config sections, it seems. But you can stuff another stage in between the first override and the original command like this if that is an option in your case:

```
[gcode_macro M104]
rename_existing: M104.1
gcode:
    { action_respond_info("In First Rename") }
    M104.1 { rawparams }

[gcode_macro M104.1]
rename_existing: M104.2
gcode:
    { action_respond_info("In Second Rename") }
    M104.2 { rawparams }
```

This will run in the order M104 --> M104.1 --> M104.2

Markus-1984, 2022-01-04:
that's what I thought, thanks a lot

Sineos, 2022-01-10:
Seems the topic is concluded. Please do not hesitate to open a new issue on https://klipper.discourse.group if needed

closed 2022-01-10 by Sineos