Klipper3d/klipper #5111: BUG: ReOpen G-code Macro parameter calls inconsistent

Markus-1984, 2022-01-05:
Hi @KevinOConnor ,

according to https://github.com/Klipper3d/klipper/issues/2356 the inconsistent parameter calls should be fixed.

I´m trying to get a T0 and T1 macro working for a manual filament change on a single extruder printer.
I´m not able to parse parameters into the macro if it´s called T0 FILAMENT=ABS for example.

[gcode_macro T0]
gcode:
    {% set filament = params.FILAMENT|default("") %}
    {% set string_output = "Please load " ~ filament ~ " Filament" %}
    #PAUSE
    { action_respond_info(rawparams ~ " (rawparams)") }
    { action_respond_info(filament ~ " (variable filament)") }
    { action_respond_info(string_output) }

Output:
10:50:45 Please load = Filament
10:50:45 = (variable filament)
10:50:45 FILAMENT=ABS (rawparams)
10:50:45 T0 FILAMENT=ABS

On the other hand it works well with:

[gcode_macro TZERO]
gcode:
    {% set filament = params.FILAMENT|default("") %}
    {% set string_output = "Please load " ~ filament ~ " Filament" %}
    #PAUSE
    { action_respond_info(rawparams ~ " (rawparams)") }
    { action_respond_info(filament ~ " (variable filament)") }
    { action_respond_info(string_output) }

Output:
10:50:53 Please load ABS Filament
10:50:53 ABS (variable filament)
10:50:53 FILAMENT=ABS (rawparams)
10:50:53 TZERO FILAMENT=ABS

Markus-1984, 2022-01-05:
I´ve found a work arround solution:

[gcode_macro T0]
gcode:
    TZERO { rawparams }

[gcode_macro TZERO]
gcode:
    {% set filament = params.FILAMENT|default("") %}
    {% set string_output = "Please load " ~ filament ~ " Filament" %}
    #PAUSE
    { action_respond_info(rawparams ~ " (rawparams)") }
    { action_respond_info(filament ~ " (variable filament)") }
    { action_respond_info(string_output) }

This is working for me now, but would be nice to be fixed

simonkuehling, 2022-01-05:
How do you generate a gcode file with commands like `T0 FILAMENT=ABS` in it? Via post-processing in your slicer?

simonkuehling, 2022-01-05:
Ok, I see... but since that is an incompatible mix of traditional style gcode commands and the extended commands from Klipper, why don't you just split that into two lines and use something like
```
T[next_extruder]
REQUEST_CHANGE FILAMENT="[filament_settings_id]"
```
in PrusaSlicer and run a set of macros like 
```
[gcode_macro T0]
gcode:
    # do nothing

[gcode_macro T1]
gcode:
    # do nothing

[gcode_macro REQUEST_CHANGE]
gcode:
{% set filament = params.FILAMENT|default("") %}
{% set string_output = "Please load " ~ filament ~ " Filament" %}
PAUSE
{ action_respond_info(string_output) }
```

This way you keep the gcode syntax clean

Markus-1984, 2022-01-05:
I understand what you mean. It´s a good and clean solution.
However, it is also just a workaround to the real problem, that macros with a number at the end could not parse parameters correct.

simonkuehling, 2022-01-05:
This is intended behavior, not a bug. The traditional syntax of gcode commands does not allow for "parameter=foobar" notation. The "extended gcode" format of commands was introduced by Klipper and cannot be mixed with the traditional syntax.

You can use your own parameters with a macro in the traditional gcode style like T0, too - if you stick to the [single character][number] syntax (Like `T0 S0`) - but you cannot pass strings to the macro this way.

The problem is not the number at the end - but if it is a *single* character followed by a number, *then* it is parsed as a traditional style command:

```
[gcode_macro TF2]
gcode:
    { action_respond_info(params.FILAMENT) }
```
executed as `TF2 FILAMENT="hello"` does work, for example.

Markus-1984, 2022-01-05:
Ok, that makes sense. Thanks for your explanaition. I´ll close it now.

closed 2022-01-05 by Markus-1984