Klipper3d/klipper #5112: set TMC2209 stealthchop threshold for sensorless homing

FotoFieber, 2022-01-06:
As discussed in https://github.com/Klipper3d/klipper/issues/2660, it is not possible to change the stealthchop_threshold mid print.

According to the Klipper Documentation, it is advised to set stealthchop_threshold to 0 for more accuracy.

I changed it but now sensorless homing doesn't work anymore.

It would be nice, if I could change the value in my homing_override:
```

[homing_override]
axes: xyz
set_position_z: 5
gcode:
    M400                  # Wait for moves to finish
    G90                   # Absolute positioning
    G0 Z10 F600           # Hop Z-Axis
    M204 S1000            # Set homing acceleration (important!)

    SET_TMC_FIELD STEPPER=stepper_x FIELD=stealthchop_threshold VALUE=999999
    SET_TMC_FIELD STEPPER=stepper_y FIELD=stealthchop_threshold VALUE=999999

    # Home Y
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.4
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.4
    G28 Y
    G0 Y{printer.toolhead.axis_maximum.y / 2} F9000

    # Home X
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=0.5
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=0.5
    G28 X

    G0 X{printer.toolhead.axis_maximum.x / 2} F9000
 
    # Restore current
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={printer.configfile.config["tmc2209 stepper_x"].run_current}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={printer.configfile.config["tmc2209 stepper_y"].run_current} 

	G0 X{printer.toolhead.axis_maximum.x / 2} Y{printer.toolhead.axis_maximum.y / 2} F3000
	
    # Rehome Z
    G28 Z
    
    # Safe Z
    G0 Z10 F600

    SET_TMC_FIELD STEPPER=stepper_x FIELD=stealthchop_threshold VALUE=0
    SET_TMC_FIELD STEPPER=stepper_y FIELD=stealthchop_threshold VALUE=0

```

FHeilmann, 2022-01-07:
`SET_TMC_FIELD` is meant to write registers in the TMC drivers, not config elements. the fields you're looking to write are `tpwmthrs` and `en_pwm_mode`. Unfortunately you can't write mm/s values to those either. If you want to do the math you can do so by checking this part of the tmc klippy code:

<https://github.com/Klipper3d/klipper/blob/master/klippy/extras/tmc.py#L517>

FotoFieber, 2022-01-11:
Sorry for my lack of know how, I can't apply the info given above.
I don't need a dynamic setting, only enable and disable.

How can I set them in [homing_override]?

a) to 999999 for sensorless homing
b) to 0 after homing

jakep82, 2022-01-11:
Klipper automatically sets the proper registers for sensorless homing regardless of whether you have stealthchop disabled. I have stealthchop disabled on my Voron and sensorless homing works exactly as expected.

jakep82, 2022-01-11:
I would recommend following this guide and using the macros listed there. 

https://www.klipper3d.org/TMC_Drivers.html#sensorless-homing

closed 2022-02-23 by github-actions[bot]