Klipper3d/klipper #5114: Input shaper with adxl345 - No module named numpy ?

StefanMeGit, 2022-01-06:
Hi, i tried to create the pngs like u mentioned in the manual, but i got errros like this:

```
pi@octopi:~/klipper $ ~/klipper/scripts/calibrate_shaper.py /tmp/resonances_y_*.csv -o /tmp/shaper_calibrate_y.png
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/calibrate_shaper.py", line 11, in <module>
    import numpy as np, matplotlib
ImportError: No module named numpy
```
But I have installed everyting correctly i guess.

```
pi@octopi:~/klipper $ ~/klippy-env/bin/pip install -v numpy
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at ht
[klippy (1).log](https://github.com/Klipper3d/klipper/files/7821821/klippy.1.log)
Using pip 20.3.4 from /home/pi/klippy-env/local/lib/python2.7/site-packages/pip (python 2.7)
Non-user install because user site-packages disabled
Created temporary directory: /tmp/pip-ephem-wheel-cache-YfY5w0
Created temporary directory: /tmp/pip-req-tracker-LuGOpq
Initialized build tracking at /tmp/pip-req-tracker-LuGOpq
Created build tracker: /tmp/pip-req-tracker-LuGOpq
Entered build tracker: /tmp/pip-req-tracker-LuGOpq
Created temporary directory: /tmp/pip-install-QWbCZu
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /home/pi/klippy-env/lib/python2.7/site-packages (1.16.6)
Removed build tracker: '/tmp/pip-req-tracker-LuGOpq'
```

```
pi@octopi:~/klipper $ sudo apt install python3-numpy python3-matplotlib
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-matplotlib is already the newest version (3.0.2-2).
python3-numpy is already the newest version (1:1.16.2-1).
The following packages were automatically installed and are no longer required:
  libjpeg8 liblua5.3-0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```
[klippy (1).log](https://github.com/Klipper3d/klipper/files/7821825/klippy.1.log)

netweaver1970, 2022-01-06:
as I recall I had to do 
`python3 pip install -v numpy`
Before it worked. 
And then everywhere (if the normal commands weren't working due to python version issues) put python3 in front. 
Maybe not the cleanest solution but it worked.

Sineos, 2022-01-06:
As you have python3 and the corresponding numpy module installed, you need to call the script via:

`python3 ~/klipper/scripts/calibrate_shaper.py /tmp/resonances_y_*.csv -o /tmp/shaper_calibrate_y.png` 

Alternatively, you can update to Klipper's latest git version, where this became the standard.

badabing2005, 2022-01-07:
I find that in many cases `klippy-env` is set to Python 2.7 whereas calibrate_shaper.py is calling for python 3
when this step is done
`~/klippy-env/bin/pip install -v numpy`
numpy is installed in python 2.7

So to get it to work one has to call
`python ~/klipper/scripts/calibrate_shaper.py /tmp/resonances_x_*.csv -o /tmp/shaper_calibrate_x.png`
where the default python is 2.7

Sineos, 2022-01-07:
You have to differentiate between these Python versions:

**klipp-env aka python venv**
This is an isolated environment that Klipper uses as his runtime environment, i.e. all internal Klipper functions draw upon this env. Still today this is Python 2.7 although endeavors are ongoing to slowly migrate to Python 3.

**OS python**
You typically have a global operating system Python installation as well. Previously some Python 2.x version, in recent distributions always some 3.x versions.
When you call a script from the command line like `~/klipper/scripts/calibrate_shaper.py` it will be executed with the global OS Python.
The header of the script, i.e. https://github.com/Klipper3d/klipper/blob/7c0559c6e62506af73d0e8f22733615705664dd5/scripts/calibrate_shaper.py#L1 tells the script which Python to use. Just recently this header information has been switched from Python 2 to Python 3.

If you still use an older Klipper (which defaults to Python 2) you can force to use Python3 by just calling it explicitly, i.e. 
`python3 ~/klipper/scripts/calibrate_shaper.py /tmp/resonances_y_*.csv -o /tmp/shaper_calibrate_y.png`

badabing2005, 2022-01-08:
@Sineos 

Everything that you say is right, but here's the gotcha
Documentation guides the users into a mixed environment usage which is the root of the issue.

https://www.klipper3d.org/Measuring_Resonances.html?h=adxl#software-installation
`~/klippy-env/bin/pip install -v numpy`
Installs numpy in Klipper Python Environment.

https://www.klipper3d.org/Measuring_Resonances.html?h=numpy#measuring-the-resonances_1
```
~/klipper/scripts/calibrate_shaper.py /tmp/resonances_x_*.csv -o /tmp/shaper_calibrate_x.png
~/klipper/scripts/calibrate_shaper.py /tmp/resonances_y_*.csv -o /tmp/shaper_calibrate_y.png
```
Uses the system Python

It would be best if both steps use one environment to avoid bad user experience.
People who understand these nuances can work around it, but many others encounter the issue and seek help.

Sineos, 2022-01-09:
> Everything that you say is right, but here's the gotcha
> Documentation guides the users into a mixed environment usage which is the root of the issue.

Sorry, this is not correct.
The venv Klipper uses is only meant to be used for Klipper internal functions 
The additionally provided scripts are meant to be used with the OS Python.

See https://www.klipper3d.org/Measuring_Resonances.html#software-installation as one example:

**Internal usage:**
```
~/klippy-env/bin/pip install -v numpy
```

**Additional scripts:**
```
sudo apt update
sudo apt install python3-numpy python3-matplotlib
```

badabing2005, 2022-01-09:
> **Additional scripts:**
> 
> ```
> sudo apt update
> sudo apt install python3-numpy python3-matplotlib
> ```

OK, good.
This must have been updated recently
To install `python3-numpy` instead of `python-numpy`
I grepped my command history for numpy to be sure
```
 history | grep numpy
 1030  ~/klippy-env/bin/pip install -v numpy
 1032  sudo apt install python-numpy python-matplotlib
 ```
 
I also see in Klipper discord channel references to `sudo apt install python-numpy python-matplotlib`

There shouldn't be any issues moving forward.
Thanks

Sineos, 2022-01-09:
Closed as the discussion seems concluded.

closed 2022-01-09 by Sineos