Klipper3d/klipper #5135: Issues with Macro gcode print_start for Cura

KC6331, 2022-01-10:
I've been trying to get a START_PRINT macro to work with Cura....If I have the Start gcode in Cura set with "START_PRINT T_BED=[first_layer_bed_temperature] T_EXTRUDER=[first_layer_temperature]", I get a t_bed error.  If I set the strart gcode in Cura to "STAR_PRINT", I get " !! Move exceeds maximum extrusion (3.338mm^2 vs 1.440mm^2)" once it starts printing the purge line

[klippy.log](https://github.com/Klipper3d/klipper/files/7836549/klippy.log)

schnello, 2022-01-10:
For your state macro:
	{% set BED_TEMP = params.**BED_TEMP**|default(60)|float %}
	{% set EXTRUDER_TEMP = params.**EXTRUDER_TEMP**|default(205)|float %}

You line in cura:
START_PRINT T_BED=[first_layer_bed_temperature] T_EXTRUDER=[first_layer_temperature]

Correct is:
START_PRINT **BED_TEMP**=[first_layer_bed_temperature] **EXTRUDER_TEMP**=[first_layer_temperature]

Sineos, 2022-01-10:
In addition to the information provided by @schnello:
The error `Move exceeds maximum extrusion` is from your priming line. Most likely caused by the `M83` command.

closed 2022-02-22 by github-actions[bot]