Klipper3d/klipper #5136: print_stats.py: Internal error when cancelling print very quickly after print start

simonkuehling, 2022-01-10:
I am running a START_PRINT macro that does some checks and should abort a print before even doing anything if some preparations (like bed mesh laveling etc.) are not done yet:

```
[gcode_macro START_PRINT]
gcode:
    {% set vars = printer.save_variables.variables %}
    {% set extruder_temp = params.EXTRUDER_TEMP|default(0)|int %}
    {% set bed_temp = params.BED_TEMP|default(0)|int %}
    {% set profile_name = printer.bed_mesh.profile_name %}

    {% if (profile_name == "") %}
        { action_respond_info("Print bed is not calibrated, aborting print!") }
        CANCEL_PRINT
    {% else %}
        ...
    {% endif %}
```

When the if-statement succeeds and `CANCEL_PRINT` is called I get the following internal error from `print_stats.py`:
```
...
Exiting SD card print (position 29691)
Unhandled exception during run
Traceback (most recent call last):
  File "/opt/klipper//klippy/klippy.py", line 201, in run
    self.reactor.run()
  File "/opt/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/opt/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/opt/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/opt/klipper/klippy/extras/virtual_sdcard.py", line 291, in work_handler
    self.print_stats.note_complete()
  File "/opt/klipper/klippy/extras/print_stats.py", line 45, in note_complete
    self._note_finish("complete")
  File "/opt/klipper/klippy/extras/print_stats.py", line 54, in _note_finish
    self.total_duration = eventtime - self.print_start_time
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
Transition to shutdown state: Unhandled exception during run
```
Looks like we can meet a race condition where `self.print_start_time` is not set yet when `_note_finish()` is called..?

I forgot to save the full log at that time - but I can reproduce the fault next time the machine is idle if needed, just let me know.

Arksine, 2022-01-10:
This issue was resolved with commit 83889ac435ee61478460adfec875fa80eb861f6d, you may need to update your version of Klipper.

simonkuehling, 2022-01-10:
oh, you're right - thanks for the hint... might have missed updating one of my systems here recently...

closed 2022-01-10 by simonkuehling