Klipper3d/klipper #5144: Speed modifier modifies retraction speed

koshisan, 2022-01-12:
I am not 100% sure, since I have been only using klipper for a short while, but I think the software isn't behaving as it should in this specific situation:

- I use firmware_retractions
- retract_speed is set to the maximum retraction speed my hardware can support (according to my understanding this is best practice)

Now, when I increase the speed while printing with M220, the retract_speed also gets increased - and goes over the maximum my hardware can support (I can tell from the noise...).

Is this working as intended? Shouldn't retraction moves be excluded, since they will always push the hardware over the limit when the parameters are set corretly in the printer.cfg?

I noticed that there is a paramter max_extrude_only_velocity - however, I am not sure what it does, since there isn't any documentation on it...

Sineos, 2022-01-12:
According to https://github.com/Klipper3d/klipper/blob/master/klippy/extras/firmware_retraction.py the firmware_retraction should not be influenced by a M220
Maybe @Lenbok could cross-check.

Lenbok, 2022-01-12:
The firmware retraction code does nothing special with the speed, so it depends on whether M220 implementation affects extrude-only commands. I don't know whether that is the case or not.

Related discussion from another firmware: https://forum.duet3d.com/topic/14418/what-types-of-move-should-the-speed-factor-affect/24

Sineos, 2022-01-12:
https://github.com/Klipper3d/klipper/blob/babb067b60e5a0b29fa8657faf01a709483b33b1/klippy/extras/firmware_retraction.py#L11
https://github.com/Klipper3d/klipper/blob/babb067b60e5a0b29fa8657faf01a709483b33b1/klippy/extras/firmware_retraction.py#L53-L61

Maybe my understanding of the code is incorrect but to me it seems that the G10 / G11 command always uses the speed as put forth in the config. How should this be affected by M220?

Sineos, 2022-01-12:
A, sry, of course: It is a percentage change in M220. So as regular GCODES are used here it it very likely that they are indeed affected by M220

Arksine, 2022-01-12:
If its desirable to ignore the speed multiplier during firmware retraction `M220 S100` could be added to the scripts in the G10/G11 handlers.  The previous multiplier would be restored with the call to `RESTORE_GCODE_STATE`.

Lenbok, 2022-01-13:
@Arksine You could do that as a workaround. It does seem (from the discussion linked above) that M220 should ignore certain types of commands, e.g.: homing, extrude-only (which would cover both the firmware retraction extension and any slicer-added software retraction), tool change, and possibly other categories.

koshisan, 2022-01-13:
> A, sry, of course: It is a percentage change in M220. So as regular GCODES are used here it it very likely that they are indeed affected by M220.
> 
> @koshisan can you check if [max_extrude_only_velocity](https://www.klipper3d.org/Config_Reference.html?h=max_extrude_only_velocity#extruder) limits it to your "max hardware allowable"

@Sineos I will need to check tomorrow, since I fried my mainboard yesterday...

Just to be sure, since the parameter isn't described: Format is mm/s like any other speed?

Sineos, 2022-01-13:
> Just to be sure, since the parameter isn't described: Format is mm/s like any other speed?

Yes, it it mm/s. Also see the documentation as per above link:
```
#max_extrude_only_velocity:
#max_extrude_only_accel:
#   Maximum velocity (in mm/s) and acceleration (in mm/s^2) of the
#   extruder motor for retractions and extrude-only moves. These
#   settings do not have any impact on normal printing moves. If not
#   specified then they are calculated to match the limit an XY
#   printing move with a cross section of 4.0*nozzle_diameter^2 would
#   have.
```

closed 2022-02-24 by github-actions[bot]