Klipper3d/klipper #5145: `TUNING_TOWER` does not work with `SET_TMC_FIELD`

Taudris, 2022-01-12:
I tried to use `TUNING_TOWER` to make a test print with different hysteresis values on my TMC5160 drivers like so:

`TUNING_TOWER COMMAND="SET_TMC_FIELD STEPPER=extruder FIELD=HEND" PARAMETER=VALUE SKIP=1 STEP_HEIGHT=10 START=5 STEP_DELTA=-1`

But I got this error about 8 seconds after submitting the command (the console shows that the `START_PRINT` macro had just finished):

`Error on 'SET_TMC_FIELD STEPPER=extruder FIELD=HEND VALUE=5.000000000': unable to parse 5.000000000`

It would be nice if `TUNING_TOWER` could work with commands that only accept integer values. Or if commands which only accept integer values could accept floats and either round the decimal portion toward 0, toward even, or floor it - but this doesn't seem like a good idea.

I assume I can hold myself over by writing a macro to convert the value; I'll try that when I have more time.

[klippy.log](https://github.com/Klipper3d/klipper/files/7856287/klippy.log)

camerony, 2022-01-21:
I was able to do this by writing a macro then having `TUNING_TOWER` call that macro.
`TUNING_TOWER COMMAND=SET_HEND PARAMETER=HEND START=0 FACTOR=.5 BAND=2`

`[gcode_macro SET_HEND]
gcode:
  {% set hnd = params.HEND|default(0)|int %}
  {action_respond_info("HEND: '%s'" % (hnd))}
  SET_TMC_FIELD STEPPER=stepper_x FIELD=hend VALUE={hnd}
  SET_TMC_FIELD STEPPER=stepper_y FIELD=hend VALUE={hnd}
`

Taudris, 2022-01-22:
That's basically what I ended up doing. I was already almost there with the `SET_TMC_FIELDS` macro you can see in my klippy.log, which allows you to supply a comma-delimited list of steppers. Once I added the `|int` conversion, I was off to the races. (Sadly, I wasn't able to improve print quality with this. Ah well.)

In case it helps anyone:
```
[gcode_macro SET_TMC_FIELDS]
gcode:
    {% set steppers = params.STEPPERS|default("stepper_x,stepper_y") %}
    {% for stepper in steppers.split(',') %}
      {% if stepper != "" %}
        SET_TMC_FIELD STEPPER={stepper} FIELD={params.FIELD} VALUE={params.VALUE|int}
      {% endif %}
    {% endfor %}
```

Then to actually use it:
`TUNING_TOWER COMMAND="SET_TMC_FIELDS STEPPERS=extruder FIELD=HEND" PARAMETER=VALUE SKIP=1 STEP_HEIGHT=10 START=5 STEP_DELTA=-1`

I'm going to close this issue. It's a *really* narrow use case.

closed 2022-01-22 by Taudris