Klipper3d/klipper #5146: Float conversion partially buggy

OXERY, 2022-01-12:
In the following Gcode, the float conversion does not work as intended. Values are then displayed with a Zero-value.
G80 MIN_X=95.9195 MAX_X=141.63 MIN_Y=93.5951 MAX_Y=116.405
`[gcode_macro G80]
gcode:
    M118 Values without conversion {params.MIN_X} {params.MAX_X} {params.MIN_Y} {params.MAX_Y}
    M118 Values as float {params.MIN_X|float} {params.MAX_X|float} {params.MIN_Y|float} {params.MAX_Y|float}`
-> Values as float is displayed as zeroes only

When executing the same macro with another name, it works as intended:
GEIGHTY MIN_X=95.9195 MAX_X=141.63 MIN_Y=93.5951 MAX_Y=116.405

`[gcode_macro GEIGHTY]
gcode:
    M118 Values without conversion {params.MIN_X} {params.MAX_X} {params.MIN_Y} {params.MAX_Y}
    M118 Values as float {params.MIN_X|float} {params.MAX_X|float} {params.MIN_Y|float} {params.MAX_Y|float}`

Of coure, I also tested `G29` ;-) Same problem.
Pls see screenshot:

[image]

Arksine, 2022-01-12:
Its invalid to use "extended" style parameters with traditional G-Codes.  For example, the following would work:
```
G80 X95.9195 R141.63 Y93.5951 S116.405
```
However, the following does not parse correctly due to the combination of a traditional G-Code (G80) with extended parameters (MIN_X=95...etc):
```
G80 MIN_X=95.9195 MAX_X=141.63 MIN_Y=93.5951 MAX_Y=116.405
```

OXERY, 2022-01-13:
Thx @Arksine

I will use a custom named macro instead!

Sineos, 2022-01-13:
Discussion seems concluded.

closed 2022-01-13 by Sineos