Klipper3d/klipper #5147: Klipper support Lerdge K Board Support ?

Witawat, 2022-01-13:
Klipper support Lerdge K Board Support ?

[image]

Sineos, 2022-01-13:
This would need some interested developer who owns this board to take up the task.
On first glance it seem not so easy, as the firmware of the Lerdge boards is encrypted. A Marlin developer has broken this encryption (see https://github.com/MarlinFirmware/Marlin/pull/18302) and the pins would be available from this PR as well.

To successfully flash Klipper firmware, I would assume that extra steps are needed

BapeHiks, 2022-01-28:
I really want to see support for this board, in lerdge-k v2.0.1 itself, the inability to use klipper on it is very disappointing

jalanjarosz, 2022-01-28:
This board uses the ARM Cortex M4, which is already supported in Klipper.  

The Arm Cortex M4 is used in the Adafruit Metro M4 and Grand Central M4 boards. 

What is the full chip number of the M4 processor?

jalanjarosz, 2022-01-28:
Interesting that the image has ARM Cortex-M4, but their web site has STM32F407ZGT

Looks like Klipper supports the stmz32f4 in some capacity.  

You might be okay with the controller, but the Lerdge display may be an issue.

jalanjarosz, 2022-01-28:
Image is from AliExpress, The mainboard has a STM32 made by ARM

[image]

Klipper does support the STM32F407 boards. 
[image]

You'd need the Lerdge schematic to map pins, unless they are silkscreened on the board.  

Good luck.

BapeHiks, 2022-01-28:
i have the board and pinout including lcd. if the task is only in the board config, then this should not be a problem, one of these days I will try to adapt

jalanjarosz, 2022-01-28:
There is an entire article on Hackaday as well as some other closed Klipper requests you should look at.  Seems the hardest part may be flashing the MCU.