Klipper3d/klipper #5148: No way to use expressions in macros without running output as command

r3Fuze, 2022-01-13:
Here's an example macro:
```
[gcode_macro NONE_TEST]
gcode:
  { [1,2].append(3) }
```

Running the macro gives `Unknown command:"NONE"` in the console since the output of the expression gets ran as a command. Jinja has an extension that provides a `do` tag that ignores the return value of expressions but it is not enabled in Klipper: https://jinja.palletsprojects.com/en/2.10.x/extensions/#expression-statement

With this extension enabled you *should* be able to run `{% do [1,2].append(3) %}` without running the output as a command.
Is there an alternative way to do this in Klipper or should this extensions maybe be activated if there are no consequences?

rdbahm, 2022-01-22:
One way to accomplish what you're asking for is to set a variable to the output of your expression. In your example, you could do:
{% set ignore = [1,2].append(3) %}

This should result in the expression being evaluated but essentially thrown away.

zellneralex, 2022-02-27:
PR https://github.com/Klipper3d/klipper/pull/5149 is waiting to be merged to solve that.

Sineos, 2022-03-25:
Seems concluded with #5149

closed 2022-03-25 by Sineos