Klipper3d/klipper #5153: [screws_tilt_adjust] with wrong desciption from "minutes" is false?

StefanMeGit, 2022-01-14:
I rescently installed a probe on my corexy and i found some maybe wrong desciption of minutes in the program. 

1 full turn = 360 degrees = 360 x 60 minutes
1 full turn = 21600 minutes 

In the example we can find 1:15 = 1 full turn and 15 minutes = 1.00069 turns? I think a calculation to minuts is a bit too much. Maybe u mean degrees instead of minutes? Or maybe 1 turn is equal to 60 minutes (which is false...)

**Please correct me if i am wrong!!**

```
Send: G28
Recv: ok
Send: SCREWS_TILT_CALCULATE
Recv: // 01:20 means 1 full turn and 20 minutes, CW=clockwise, CCW=counter-clockwise
Recv: // front left screw (base) : x=-5.0, y=30.0, z=2.48750
Recv: // front right screw : x=155.0, y=30.0, z=2.36000 : adjust CW 01:15
Recv: // rear right screw : y=155.0, y=190.0, z=2.71500 : adjust CCW 00:50
Recv: // read left screw : x=-5.0, y=190.0, z=2.47250 : adjust CW 00:02
Recv: ok
```

[klippy (2).log](https://github.com/Klipper3d/klipper/files/7869847/klippy.2.log)

Sineos, 2022-01-14:
You are thinking too complicated. It is a "picturesque" value referring to the minute-hand on an  analogue watch. One full round of the minute-hand is 1h.
Or maybe you are too young for analogue watches 😉 (just kidding)

StefanMeGit, 2022-01-14:
Ahhhhhhh OMG *facepalm*! Sorry. 

MAYBE you can add this information to the wiki. Maybere there are some other supid guys like me. Also a hint about what is clockwise and what is counter clockwise (watch from the top or bottom, crew or nut... ect.).

Thanks anywa, The screws_tilt_adjust works like a charm!

closed 2022-01-14 by StefanMeGit