Klipper3d/klipper #5164: Klipper fails to start with a nozzle_diameter of 1.8 and a filament_diameter of 1.75

deece, 2022-01-16:
With the following settings added to support the CHT 1.8mm nozzle on a 1.75mm filament machine:
```
nozzle_diameter: 1.75
filament_diameter: 1.75
```

we get the following error:
```
!! Option 'filament_diameter' in section 'extruder' must have minimum of 1.8
```

Current workaround is to use:
```
nozzle_diameter: 1.75
filament_diameter: 1.75
```

I believe that with the advent of mixing nozzles such at the Bondtech CHT, this check should be relaxed.

deece, 2022-02-21:
Bump to keep the bot happy

sjlee001, 2022-02-25:
Hi deece,

Just go to the file ~/klipper/klippy/kinematics/extruder.py and modify the line 22 from:

        filament_diameter = config.getfloat(
            'filament_diameter', minval=self.nozzle_diameter)

to:

        filament_diameter = config.getfloat(
            'filament_diameter', minval=1.75)

or whatever diameter you want.

ld3300, 2022-02-27:
As an additional note, the nozzle diameter in the klipper config is only used to set the maximum extrude cross section, it has no other effect on printing.  You can manually set the cross section value and just not worry about the nozzle size.

dewi-ny-je, 2022-03-03:
> Hi deece,
>
> Just go to the file ~/klipper/klippy/kinematics/extruder.py and modify the line 22 from:
>
> ```
>     filament_diameter = config.getfloat(
>         'filament_diameter', [...]

Since you know how to modify the source code, would you mind making a pull request where the "minval" is set to 1.15 times "self.nozzle_diameter" and commit it? it's a reasonable value which, due to recent developments in nozzles ("CHT" basically), could be useful to several users.

1.15 times means up slightly more than 2 mm nozzle for 1.75 mm filament, which covers the current 1.8 mm and potential refinements of that, maybe going up to 2 mm.

lunatic1972, 2022-03-04:
I had the exact same issue with CHT nozzles but this post explains it in more detail

https://github.com/Klipper3d/klipper/issues/4985

Sineos, 2022-03-27:
Discussion seems to have concluded. In case of a follow up lease open a new topic on https://klipper.discourse.group

closed 2022-03-27 by Sineos