Klipper3d/klipper #5165: Option 'set_servo servo' is not valid in section 'probe'

MsnSazabi, 2022-01-16:
Hello,

Trying to setup servo activated probe:
[servo BFPTouch]
pin: PA1
initial_angle: 0
maximum_servo_angle = 90
minimum_pulse_width = 0.001
maximum_pulse_width = 0.002

[probe]
pin: ^PC14
x_offset: 0.0
y_offset: 25.0
z_offset: 1
speed: 5.0
samples: 2
activate_gcode:
SET_SERVO SERVO=BFPTouch ANGLE=0
G4 P300
deactivate_gcode:
SET_SERVO SERVO=BFPTouch ANGLE=100

And keep getting this error: Option 'set_servo servo' is not valid in section 'probe'. As well as 
File contains parsing errors: /home/pi/klipper_config/printer.cfg
[line 121]: 'G4 P300\n'

I'm getting this error for every gcode_macro I'm trying to make. The only section it worked for me is homing_override. But I need to activate probe for bed leveling.

[klippy (1).zip](https://github.com/Klipper3d/klipper/files/7876855/klippy.1.zip)

Sineos, 2022-01-16:
Your config is:
```
[servo BFPTouch]
pin: PA1
initial_angle: 0
maximum_servo_angle = 90
minimum_pulse_width = 0.001
maximum_pulse_width = 0.002

[probe]
pin: ^PC14
x_offset: 0.0
y_offset: 25.0
z_offset: 1
speed: 5.0
samples: 2
```

The rest:
```
# activate_gcode: <--- just a comment /heading
SET_SERVO SERVO=BFPTouch ANGLE=0
G4 P300
```
and
```
# deactivate_gcode:  <--- just a comment /heading
SET_SERVO SERVO=BFPTouch ANGLE=100
```

Are the commands you would use to deploy or retract the probe

zellneralex, 2022-02-15:
The lines are not allowed to be at line start change it to:
`
[servo BFPTouch]
pin: PA1
initial_angle: 0
maximum_servo_angle = 90
minimum_pulse_width = 0.001
maximum_pulse_width = 0.002

[probe]
pin: ^PC14
x_offset: 0.0
y_offset: 25.0
z_offset: 1
speed: 5.0
samples: 2
activate_gcode:
  SET_SERVO SERVO=BFPTouch ANGLE=0
  G4 P300
deactivate_gcode:
  SET_SERVO SERVO=BFPTouch ANGLE=100`

Sineos, 2022-03-08:
Topic seems concluded.
If there is the need to continue, please have a look at: https://klipper.discourse.group/t/servo-deployed-abl-z-homing-issue-and-fix/265

closed 2022-03-08 by Sineos