Klipper3d/klipper #5168: !! Move exceeds maximum extrusion (4.789mm^2 vs 0.640mm^2)

Reybad776, 2022-01-16:
every time I want to print something I get this error after the printer finishes heating hotend and the print bed, I didn't have this problem before but when I changed the printer profile in the ideamaker slicer I started getting this problem

[klippy.log](https://github.com/Klipper3d/klipper/files/7877856/klippy.log)

Sineos, 2022-01-17:
> This "max_extrude_cross_section: 1.0" is too low by you. :)

Without special requirements, there is no need to set `max_extrude_cross_section` at all. The default value is calculated and typically fits every normal printing task (for a 0.4 nozzle and 1.75 filament, 0.64 is default). 
Overall this setting is meant to prevent damage to the extruder system by requesting too high values on short distances. It should **not** be modified for regular printing scenarios. 

If such errors pop up, usually something is funky in start GCODE like requesting an insane amount during the purge line or having misplaced M83.
Also it may help to do the purging in relative mode (G91) instead of absolute (G90)

Sineos, 2022-02-18:
Topic seems to be concluded. For further discussions please do not hesitate to open a new topic on https://klipper.discourse.group/.

closed 2022-02-18 by Sineos