Klipper3d/klipper #5170: Voron 2.4 Loss of z end stop location

tempestaz, 2022-01-17:
I had to abort a print, via octoprint, to resolve some printer related belt issues. Once resolved I restarted the homing process. I was able to successfully home XY. When I went to home Z, instead of it going to the z end stop position, the head moved to the upper left corner (approx 300,300) of the build plate and crashed.

Upon a clean restart of everything, z homing works fine. 

log file is attached
[klippy.log](https://github.com/Klipper3d/klipper/files/7884474/klippy.log)

Sineos, 2022-01-18:
If you can reproduce the problem, please provide the necessary steps to do so. Otherwise I recommend to post on https://klipper.discourse.group if other people have experienced a similar issue

tempestaz, 2022-01-20:
I tired numerous times to reproduce this yesterday. No luck. Should I be able to determine a set of steps that reproduce this I will open a new issue. In the interim, I'll post to discord.

Thanks!

Sineos, 2022-01-20:
Ref to https://klipper.discourse.group/t/head-crash-on-z-homing-voron-2-4/1837

closed 2022-01-20 by Sineos