Klipper3d/klipper #5176: 5 z axis motors, but only used 1 z axis motors to run for printing

daxzero, 2022-01-20:
I have 5 z axis motors, if I want to run only 1 z axis motor when printing. The others 4 z axis motors will be doing for bed levelling only, after that will stay idle. Can this be done?

Sineos, 2022-01-20:
This sounds like a very special case and to my knowledge it is not possible.
Also the question is, how this should work generally:
* 1 single stepper shall move a bed that is being held by an additional 4 steppers? 
* Apart from firmware topics, this sounds very weird mechanically --> massively over-constrained system 
* When these stepper are enabled they will block movement up to their holding torque --> generally when one Z stepper is enabled all are enabled, although they could be disabled manually: https://www.klipper3d.org/G-Codes.html?#manual-stepper-commands

Probably I do not fully understand the concept but at first glance this sounds problematic

Generally, please open such a discussion on https://klipper.discourse.group

daxzero, 2022-01-20:
ok. i already open the discussion in the Klipper group. We can continue there.

closed 2022-01-20 by Sineos

daxzero, 2022-01-20:
here is the link: https://klipper.discourse.group/t/5-z-axis-motors-but-only-used-1-z-axis-motors-to-run-for-printing/1841