Klipper3d/klipper #5178: Ender 3v2 Klipper scrapes and grinds through print/infill.... I'm lost....

Mikei386, 2022-01-20:
Hi there,

I’m using Klipper on my Ender 3 v2 for about 4 months now, but there is one thing I can’t get quite right.
Here are my details:

System:

Ender 3 v2:
- Stock except for the following upgrades:
* Aluminum Extruder
* Capricorn Bowden Tube
* Orange bed springs
* BL-Touch Sensor
* Dual-Z-Axis kit
* Sunon Fans

Firmware:
* Klipper v0.10.0-223-g4c8d24ae
* Fluidd v1.16.2
* Moonraker v0.7.1-285-g505c1bf
* Klipperscreen v0.1.6-194  
* Config: [printer_cfg.txt](https://github.com/Klipper3d/klipper/files/7904707/printer_cfg.txt)
(All installed via the kiauh script)

Slicer:
* Cura

Issue:
On every print I do the first (slow) layers are pretty nice but then, most of the time when the printer is doing infill, the nozzle starts grinding and scraping at every layer change on it's way back right through the infill with horribly noises. Even at the normal speed ( 50mm/sec).

When using higher speeds (80mm/sec and above) it gets so aggressive that the infill gets torn. 
At some point it actually ripped the print right of the bed.
However the finished print looks ok. Top and Bottom Surfaces are good and the walls look nice.

I’m using a BL-Touch. I also have better bed-springs installed. I use 3D-Lac adhesive.
I calibrated my e-steps multiple times… It’s spot on….
I played with the z-offset.
I even bought a dual z-axis kit to prevent gantry sagging, but no change.
I played with the temperature (just using PLA).
I bought a new glass bed.

The only way to print “normal” without scraping and grinding noises is to use z-hop.

When reverting back to the Marlin firmware it prints the exact same g-code perfectly fine…
I would hate to leave Klipper behind and go back to Marlin just to get the normal print operation back, but as of now I'm totally lost.

Sineos, 2022-01-20:
Please refrain from cross posting

closed 2022-01-20 by Sineos

Mikei386, 2022-01-20:
Sorry but I have no idea what you are talking about?
This the first time I'm posting about this problem.

If there is any thread with the same problem (which I didn't find) it would be nice to have a link.

Otherwise I don't know what you mean. I didn't post about my problem anywhere else other than here at this time
This is my first post ever on the Klipper topic.

Mikei386, 2022-01-20:
I guess you think I postet this elsewhere because I use listing used on a discourse group threat?
I just used the type of listing since it seems the neatest to me.

@Sineos

Sineos, 2022-01-20:
Your post looks nearly identical to https://klipper.discourse.group/t/ender-3-v2-help/1798
My apologies if it was not you.
Anyway, such request is much better placed on Discourse since it is the target to use Github only for development purposes.

Mikei386, 2022-01-20:
Thank you for your reply.
I honestly appreciate it.

I used the same layout as that post since I thought it would be the cleanest way to post all the details. 

I will follow your advice and post it at the Discourse page.