Klipper3d/klipper #5180: FLSUN QQS-PRO Option 'uart_pin' in section 'tmc2209 stepper_a' must be specified

Xcat2008, 2022-01-20:
Hello, thank you in advance.
I'm stuck on this error message :/ I have 2208 drivers, dunno if is that the main problem on my QQS-Pro. If i apply your printer config Stock, everything works, less the HOME ALL but maybe is backwards. When applying your config with 2209 got that error... I do not know what do more.. Already spend a lot of hours truing to figure out, i know from the message i need to set a number or something for UART connection, but dunno what set there. Can you help? Got also one video, because my english is not that great. Thanks https://youtu.be/vzhqECSI52
[image]
[image]
o

Sineos, 2022-01-20:
Please post your support request on https://klipper.discourse.group together with your `klippy.log` file.

Xcat2008, 2022-01-20:
Hey. Think i figure out my problem, thank you. The options was almost in the last of the config xD And now, since i have drivers 2208 V2.0 need to solder 2 pins from the drivers, to be able Klipper comunicate, before had only 1 pin connected.

Sineos, 2022-02-18:
Topic seems to be concluded. For further discussions please do not hesitate to open a new topic on https://klipper.discourse.group/.