Klipper3d/klipper #5182: Event triggers

azytzn, 2022-01-22:
Hi all. I don't know if anyone has mentioned about this before but if klipper have trigers like OnStart, OnPrintStart, OnPrintEnd... it would be very nice.

For example:

[Action OnStart] 
home 
preheat
level bed
turnonled_etc..
#end

[Action OnPrintProgress]
ChangeLedcolor_etc..
#end

[Action OnCommand]
command: Z_TILT 
todo: gcodemacroname
#end

(Please excuse my english.) Thanks all

yzheka, 2022-01-30:
You can do that with macro. For example create macro OnStart (actually name can be any) and call it in your slicer's start gcode. OnPrintProgress can be done by extending M73, like OnCommand by defining Z_TILT macro.

Sineos, 2022-02-18:
Topic seems to be concluded. For further discussions please do not hesitate to open a new topic on https://klipper.discourse.group.

closed 2022-02-18 by Sineos