Klipper3d/klipper #5186: multiple printers / Same model / Same Board but can control only one of them at a time

jfle, 2022-01-23:
Hello  I have an issue on my raspberry setup. I have 2 3D Printers they are identical models. They use the CH340 serial to usb. 
(Flsun Q5)
If i run:$ ls -l /sys/bus/usb-serial/devices
Then i get:
total 0
lrwxrwxrwx 1 root root 0 Jan 23 01:53 ttyUSB0 -> ../../../devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/ttyUSB0
lrwxrwxrwx 1 root root 0 Jan 23 01:53 ttyUSB1 -> ../../../devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/ttyUSB1

if i run: lsusb

Then i get the following:
**Bus 001 Device 012: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 023: ID 1a86:7523 QinHeng Electronics CH340 serial converter**
Bus 001 Device 004: ID 07d1:3c0a D-Link System DWA-140 RangeBooster N Adapter(rev.B2) [Ralink RT3072]
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

If i finally run: ls /dev/serial/by-id/*
Then i get only 1 serial port as bellow
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

So the problem is that depending which printer was the last one to plug in is the one that i can control (Klipper / Fluidd) it seems that i have 2 usb devices but only 1 serial port mapped and it is mapped to both of them.

jakep82, 2022-01-23:
Use by-path instead of by-id. Please direct future help requests to Discord or Discourse.

Sineos, 2022-02-18:
Topic seems to be concluded. For further discussions please do not hesitate to open a new topic on https://klipper.discourse.group.

closed 2022-02-18 by Sineos