Klipper3d/klipper #5189: BUG?: Chamber Heater with DS18B20 Temperature Sensor generates "MCU 'mcu' shutdown: Scheduled digital out event will exceed max_duration" failure

Markus-1984, 2022-01-23:
Hi,

I´m not able to get the Chamber Heater working with the DS18B20 Sensor.
If I set the DS18B20 only as a temperature sensor it works well:

[temperature_sensor chamber]

gcode_id: CHAMBER_TEMP
sensor_type: DS18B20
sensor_mcu: rpi
serial_no: 28-3c01f095187a
ds18_report_time: 2

If I set it in the generic heater section like this:

[heater_generic chamber]
heater_pin: PA3
sensor_type: DS18B20
serial_no: 28-3c01f095187a
ds18_report_time: 2
sensor_mcu: rpi
min_temp: 0
max_temp: 100
#control: watermark
#control: pid
#pid_kp: 100.0
#pid_ki: 1.0
#pid_kd: 1.0
gcode_id: CHAMBER

 It will generate a "MCU 'mcu' shutdown: Scheduled digital out event will exceed max_duration" error

As I started with the DS18B20 sensor, the default value for the ds18_report_time: 1 was to low, and generates the same error.
So I´ve increased it to 2 and so it works well.

Referring to https://github.com/Klipper3d/klipper/issues/1641#issuecomment-494381609 you wrote that the temp sensors are sampled every 0.3 seconds. Is it possible that klipper ignores the ds18_report_time: 2 in the generic_heater section and this would be the reason for this error?

You can see it in the attached LOG that the DS18B20 is the reason for the error, and it´s the same error if I set the ds18_report_time to 1.

[klippy.log](https://github.com/Klipper3d/klipper/files/7921272/klippy.log)

Sineos, 2022-01-23:
Default for `ds18_report_time` is but the allowed range is 1 to 3.
Otherwise maybe @theopensourcerer or @akshaal could provide some insight

Markus-1984, 2022-01-23:
Good to know, but for me 1 was not working. Only after I have increased it to 2, the sensor works as a temperature sensor. But not as a sensor for a heater. That's the reason why I think the generic_heater will ignore the statement for the longer sample time.

charredchar, 2022-01-25:
I am having a similar issue and I was able to verify it is caused by the DS18B20. I removed my `heater_bed` options and added that thermistor to the `generic_heater` for testing as I have no other options for temperature sensors. In this case it was able to trigger the `generic_heater` and function properly.

I've attempted many different options to resolve this, the best I could get is a longer delay between the heater turning on and the error by increasing the `ds18_report_time` as much as 10 seconds. This is obviously not desired as a longer report time can cause other issues.

[klippy (1).log](https://github.com/Klipper3d/klipper/files/7935857/klippy.1.log)

charredchar, 2022-01-26:
I wanted to add that I did one more test to find any further issues. I added it as a `temperature_fan`, using the same DS18B20 sensor and it had no issues. Of course this was heating when it wanted to cool, but it did not throw any errors for 2 minutes before I turned it off. All settings were the same for `generic_heater`.

theopensourcerer, 2022-01-27:
@charredchar I had seen this issue and I was one of the instigators of this module, but unfortunately I haven't turned on my printer for a few months now and probably won't have a lot time to investigate in the near future. 

You latest comment was interesting... I have only ever used the DS18B20 as a sensor or a temperature_fan, never as a heater / in a heater_generic setup and I haven't ever experienced any problems with it as a fan/sensor. 

So maybe there is a difference in the way the module interacts when configured as a heater - that might be a good avenue to follow in the short term?

One other thing I noticed in your log is that you are using the same device (28-3c01f095187a) in two separate configurations, once as a heater and once as a temperature sensor. I am not sure if that is even supposed to be allowed ;-) i.e. If you comment out either one of these two configs, does the other run without issue?

charredchar, 2022-01-27:
> So maybe there is a difference in the way the module interacts when configured as a heater - that might be a good avenue to follow in the short term?

Thanks for the response, I am hoping to try anything as I just moved from OctoPrint and this is the only thing keeping me from printing. It could very well be the case the module interacts differently for the heaters, unfortunately I do not have the skill to dig into how or why and I am relying on support for a fix. I am willing to test though, even if it is just a patch until a real fix can be done.

I've said this elsewhere, if the DS18B20 just can't be used as a heater thermostat that'll be fine by me as long as we can come up with some other option to use a temperature sensors off the Pi to run a heater since my Printer MCU is limited on thermistor inputs. If there is a reason it can't be used for a heater it should probably be noted in the docs so others don't run into the same issue.

closed 2022-03-11 by github-actions[bot]

Fjbarchesky, 2022-04-26:
I may have the same issue as mentioned in the original post. Any help would be appreciated... 

I have tried changing the time durations but either get the max_duration error or "MCU 'rpi' shutdown: Missed scheduling of next digital out event. This is generally indicative of an intermittent communication failure between micro-controller and host." 

[klippy.log](https://github.com/Klipper3d/klipper/files/8558954/klippy.log)

dmaxwellNEU, 2022-10-12:
> Thanks for the response, I am hoping to try anything as I just moved from OctoPrint and this is the only thing keeping me from printing. It could very well be the case the module interacts [...]

@theopensourcerer @charredchar do either of you have updates or developments on this? The DS18B20 sensor and generic_heater and/or rpi gpio vs pwm pins control issues are a big mystery. Where is this "module" that was mentioned for generic_heater? If copied from an extruder heater, might there be a leftover requirement in the code somewhere that is looking for a pwm pin and/or a thermistor not a DS18B20? 🤔

charredchar, 2022-10-12:
I ended up just reusing an old thermistor I had laying around and an analog pin on a secondary MCU. Even with this method if I leave any of the DS18B20 sensors enabled I get the same error eventually. There is a PR about this, I eventually just gave up on them and I highly doubt it'll ever work correctly.

dmaxwellNEU, 2022-10-12:
> 

Thanks! So are you basically saying to abandon ds18b20 sensors across the board?

dmaxwellNEU, 2022-10-12:
> I ended up just reusing an old thermistor I had laying around and an analog pin on a secondary MCU. Even with this method if I leave any of the DS18B20 sensors enabled I get the same error [...]

Thanks! Are you basically saying to abandon DS18B20 sensors across the board?