Klipper3d/klipper #5190: [FR] shared_stepper for tilting carriages

DrumClock, 2022-01-24:
https://github.com/Klipper3d/klipper/blob/28b568a6ace40af5f92383d23797409f4c1bcce3/config/sample-multi-extruder.cfg#L53-L58

A Switching Extruder is a dual extruder that uses a single stepper motor to drive two filaments,
but only one at a time. The servo is used to switch the side of the extruder that will drive the filament. 
The E motor also reverses direction for the second filament. 

### TEST 1:

```
[duplicate_pin_override]
pins: PB5, PB6, PC12

[extruder]               ### stepper motor E0
step_pin: PB5
dir_pin: PB6            ### normal rotation fot T0 (CW)
enable_pin: !PC12
...

[extruder1]             ### stepper motor E0
step_pin: PB5
dir_pin: !PB6          ### reverse rotation for T1 (CCW)
enable_pin: !PC12
....
```

I wanted to use this setting for toolhead, where two hotends (T0 / T1) use same stepper motor (E0).
When using this configurations the following problem occurs:

ACTIVATE_EXTRUDER EXTRUDER = extruder  
G1 E10 >>>> direction rotation of stepper motor E0 is OK (CW)

ACTIVATE_EXTRUDER EXTRUDER = extruder1 
G1 E10 >>>> direction rotation of stepper motor E0 is OK (CCW)

Any further change to ACTIVATE_EXTRUDER EXTRUDER no longer changes the direction of rotation 
of the motor and remains after the second change from G28

### TEST 2:

As a temporary solution, I used the [extruder] setting for T0 and [heater_generic t1] for T1.

It is necessary to G1 modification was required:

```
[gcode_macro G1]
rename_existing: G1.1
gcode:
    # E-rotation revers for HotEnd 1/T1)
    {% set user_vars = printer["gcode_macro VARIABLE"] %}
    {% set emultiplier = -1.0 if user_vars.active_tool == 1 else 1.0 %}
    {% set _E = params.E|default(0)|float * emultiplier %}
    # Get parameters
    {% set X = '' if 'X' not in params else " X%s" % params.X %}
    {% set Y = '' if 'Y' not in params else " Y%s" % params.Y %}
    {% set Z = '' if 'Z' not in params else " Z%s" % params.Z %}
    {% set E = '' if 'E' not in params else " E%s" % _E %}
    {% set F = '' if 'F' not in params else " F%s" % params.F %}
    G1.1{"%s%s%s%s%s" % (X,Y,Z,E,F)}
```

Unfortunately, it is necessary to set min_extrude_temp: 0 so that printing only works with the T1 tool when the T0 hotend is off.
Of course no configurations of type pressure_advance, max_extrude_cross_section and others which are in section [extruder] are possible for T1.

### A POSSIBLE SOLUTION

Add an option in the section for [extruder 1] this:

```
[extruder1]

shared_stepper:
#   If this extruder uses the same stepperr already defined for another
#   extruder then place the name of that extruder here. 
#   The default is to not reuse an existing extruder.

reverse_shared_stepper: 
#  If the shared extrude needs reversing rotation, set to 1. 
#  The default setting is 0

```

vladbabii, 2022-01-29:
A different temporary solution could be to override G0 and G1 with macros and put an if extruder=0 then move in the normal direction and if extruder=1 then move in the opposite direction

DrumClock, 2022-01-30:
Hi @vladbabii 
**YES** - I used override G0 and G1 with macros in **TEST 2**
**NO** - section `[extruder]` and `[extruder1]` cannot be defined because  **that uses a single stepper motor to drive two filaments**

Must be defined `[extruder]` setting for T0 and `[heater_generic t1]` for T1. 
 However, this does not allow the use of the following functions, for example: 
`min_extrude_temp`  `pressure_advance`  `max_extrude_cross_section`

Tircown, 2022-01-30:
#5210 provides a robust solution.

Sineos, 2022-02-18:
Topic seems to be concluded. For further discussions please do not hesitate to open a new topic on https://klipper.discourse.group.

closed 2022-02-18 by Sineos

DrumClock, 2022-02-18:
Hi @Sineos , the topic is solved using # 5210 only half  ..... you  ** still can't share one stepper motor for two extruders **

DrumClock, 2022-02-18:
> I'll try **TEST 1** above with the same `step_pin, dir_pin` and `nable_pin` and
> I solve the reversal of rotation with `SET_EXTRUDER_ROTATION_DISTANCE`

### so unfortunately ... it doesn't work

DrumClock, 2022-02-18:
Hi @KevinOConnor 
this looks functional
I used some unconnected pins for the` [extruder1]` configuration.

maybe it would be enough not to enter pins or dummy
(in this case it is not necessary to define TMC stepper driver configuration)
```
step_pin: dummy
dir_pin: dummy
enable_pin: dummy
```

**I will test tomorrow on real 3D printing**

```
[gcode_macro SET_HOTEND]
description: Set HotEnd: SET_HOTEND E=[0|1]
gcode:
  
    {% set hotend = params.E|int %}
  
  {% set default_servo = (1,1)[hotend] %}   
  {% set dafault_extruder =('extruder','extruder1')[hotend] %}
  {% set sync_extruder =('extruder1','extruder')[hotend] %} 
  {% set dafault_rotation = printer.configfile.settings[dafault_extruder].rotation_distance * ((1,-1)[hotend]) %}
   

  #### Tilt the toolhead 
  {% if hotend in [0,1] %}     
     ACTIVATE_EXTRUDER EXTRUDER={dafault_extruder}
     SET_EXTRUDER_ROTATION_DISTANCE EXTRUDER={dafault_extruder} DISTANCE={dafault_rotation}
     SET_EXTRUDER_ROTATION_DISTANCE EXTRUDER={sync_extruder} DISTANCE={dafault_rotation}
 
   {% if  hotend == 0 %}
     SYNC_EXTRUDER_MOTION EXTRUDER={dafault_extruder} MOTION_QUEUE={dafault_extruder}     
     SYNC_EXTRUDER_MOTION EXTRUDER={sync_extruder} MOTION_QUEUE={sync_extruder} 

     SET_SERVO SERVO={default_servo} angle=25     # position for HotEnd 0

   {% elif  hotend == 1 %}
     SYNC_EXTRUDER_MOTION EXTRUDER={sync_extruder} MOTION_QUEUE={dafault_extruder}
     SYNC_EXTRUDER_MOTION EXTRUDER={dafault_extruder} MOTION_QUEUE={sync_extruder}

     SET_SERVO SERVO={default_servo} angle=155     # position for HotEnd 1

   {% endif %}

     G4 P{user_vars.delay_1}  
     SET_GCODE_VARIABLE MACRO=VARIABLE VARIABLE=active_tool VALUE={hotend}   
  {% else %}
      RESPOND TYPE="error" MSG="Params E="{hotend} " not available for define HotEnd. "
  {% endif %}
 
  #### End of switching 

```

DrumClock, 2022-02-19:
**so I simplified the macro and I go for 3D printing** 

```
[gcode_macro SET_HOTEND]
description: Set HotEnd: SET_HOTEND E=[0|1]
gcode:
  
  {% set user_vars = printer["gcode_macro VARIABLE"] %}
  {% set hotend = params.E|default(user_vars.active_tool)|int %}
  
  {% set default_servo = (1,1)[hotend] %}  
  {% set default_angel = (user_vars.angle_e0, user_vars.angle_e1)[hotend] %} 

  {% set default_extruder =('extruder','extruder1')[hotend] %}
  {% set sync_extruder =('extruder','extruder')[hotend] %} 
  {% set dafault_rotation = printer.configfile.settings[default_extruder].rotation_distance  * ((1,-1)[hotend]) %}
   

  #### Tilt the toolhead 

  {% if hotend in [0,1] %} 

  #### set active extruder  
     ACTIVATE_EXTRUDER EXTRUDER={default_extruder}    
     SET_EXTRUDER_ROTATION_DISTANCE EXTRUDER={sync_extruder} DISTANCE={dafault_rotation}
     SYNC_EXTRUDER_MOTION EXTRUDER={sync_extruder} MOTION_QUEUE={default_extruder}
   
  #### set servo position for HotEnd 
     SET_SERVO SERVO={default_servo} angle={default_angel}
     G4 P{user_vars.delay_1} 
 
  #### save variable
     SET_GCODE_VARIABLE MACRO=VARIABLE VARIABLE=active_tool VALUE={hotend}   

  {% else %}
      RESPOND TYPE="error" MSG="Params E="{hotend} " not available for define HotEnd. "
  {% endif %}
 
  #### End of switching 

```

DrumClock, 2022-02-19:
### Printing is OK

Terminal:
```
Send: set_hotend e=1
Recv: // Activating extruder extruder1
Recv: // Extruder 'extruder' rotation distance set to -21.293000
Recv: // Extruder stepper now syncing with 'extruder1'

Recv: ok
Send: set_hotend e=0
Recv: // Activating extruder extruder
Recv: // Extruder 'extruder' rotation distance set to 21.293000
Recv: // Extruder stepper now syncing with 'extruder'

```

### Conclusion:
If it is possible to **omit the definition (or not enter)** in the `[extruder]` section` step_pin, dir_pin, enable_pin` it will work

Tircown, 2022-02-19:
Just a thought, an universal solution would be to split extruders and extruder steppers into the config too.
One can define any stepper_e section needed, `[stepper_e]`, `[stepper_e1]`, ..., and associate them in the `[extruder]` section throught the parameter `extruder_stepper: stepper_e`. `[extruder]` and `[extruder1]` sections can use the same extruder_stepper value. 
As backward compatibility one can still use the step, dir and enable pins in the extruder section.
The reverse direction can be accomplished in the activation macro or another solution would be to override or keep the rotation_distance in the extruder section (with a negative value for extruder1).

Config splitted:
```[stepper_e]
step_pin:
dir_pin:
enable_pin:
microsteps:
rotation_distance:
rotation_distance:
#full_steps_per_rotation:
#gear_ratio:

[tmc2209 stepper_e]
...

[extruder]
extruder_stepper: stepper_e
nozzle_diameter:
filament_diameter:
#max_extrude_cross_section:
#instantaneous_corner_velocity: 1.000
#max_extrude_only_distance: 50.0
#max_extrude_only_velocity:
#max_extrude_only_accel:
#pressure_advance: 0.0
#pressure_advance_smooth_time: 0.040
heater_pin:
#max_power: 1.0
sensor_type:
sensor_pin:
#pullup_resistor: 4700
#smooth_time: 1.0
control:
pid_Kp:
pid_Ki:
pid_Kd:
#max_delta: 2.0
#pwm_cycle_time: 0.100
#min_extrude_temp: 170
min_temp:
max_temp:```

DrumClock, 2022-02-19:
Yes, good idea to split **extruder** and **stepper**.

DrumClock, 2022-02-20:
Hi, or for backward compatibility, add the <extruder_stepper:> option to the extruder section and leave it up to the user to define pins or stepper.

KevinOConnor, 2022-02-21:
FYI - I created PR #5294 .

It's possible to separate out steppers from extruders in the general case, but I fear that will seem "pedantic" to most users - as the vast majority of users have a single hotend and a single extruder motor.  So, I think it makes sense to use a single config section for that common case.

Separately, note that pressure advance is currently always specified for the _stepper_ - not the extruder motion queue.  So, both before and after PR #5294 it would be necessary for `Tx` macros to explicitly manage pressure advance if a single stepper is reconfigured to push a different type of filament.

-Kevin

DrumClock, 2022-02-22:
Yes, I agree with you on both points.

I'm ill as soon as I'm fit I'll test the PR5294