Klipper3d/klipper #5195: BL-Touch works properly on G28 however the probe does not deploy for bed mesh

MDietz109, 2022-01-26:
I went through the BL touch debugging and everything works as expected. Whenever I home the printer the BL-Touch deploys and properly homes the z. However when I initiate the BED_MESH_CALIBRATE the print head moves down but the BL-Touch is stowed and BED_MESH_CALIBRATE does not deploy the pin. I have to send M112 to prevent a collision. I apologize if I am missing something simple. Help would be greatly appreciated 

[KlipperLog.txt](https://github.com/Klipper3d/klipper/files/7940004/KlipperLog.txt)

[printer confg copy.txt](https://github.com/Klipper3d/klipper/files/7940027/printer.confg.copy.txt)

MDietz109, 2022-01-26:
I should also mention which I forgot the BL-Touch is a Antclabs V3.1 with a Creality V4.2.2 board on the Ender 3 V2.

Sineos, 2022-01-26:
As GitHub shall mainly be used to target development relevant topics, please post your support request on https://klipper.discourse.group.

Try removing `probe_with_touch_mode = true`

closed 2022-01-28 by github-actions[bot]